outdoorbits/little-backup-box #5: Was working, then just stopped

ElDuderino80, 2017-05-16:
Raspi Zero W, w/ stacking USB hat from makerbot. MicroSD card reader and sandisk USB 3 flash drive.

For a while backups were fine, then it just stopped working. I followed the directions and inserted the destination first (sandisk) then the reader. It just flashes and nothing ever happens. The card ID was written to the flash drive, but nothing else. (the flash drive only illuminates activity when I first put it in, then fails to again) If I pull the sandisk drive, the pi shuts down. Is there a certain amount of time I need to wait for the pi to book? Usually I give it about a minute, does it need more?

Without feedback of any sort, it's really difficult to know what's going on with the script, such as when the pi has completed the boot sequence for example. I saw another request for some kind of board LED status light, and I think that would be helpful.

ElDuderino80, 2017-05-17:
I tried another USB thumb drive, and this time it wrote the folder structure from the SD card, but no actual files.

dmpop, 2017-05-17:
Thank you for the detailed report. You are right, without any feedback beyond the defined LED sequences, troubleshooting is almost impossible. I plan to implement some kind of logging/debugging functionality. But until then, I'm afraid that your only option is to try to reinstall Little Backup Box from scratch.

ElDuderino80, 2017-05-17:
I reinstalled it, but I have the same problem. I went as far as deleting the folder off the pi, and trying the install again. Same result. Any other thoughts on how to get it working?

ElDuderino80, 2017-05-18:
Further follow up, after reinstall I'm in a worse place. It never even sees the SD card when I SSH in and used df -h, which previously showed it. After doing nothing for a few seconds, the pi shuts down. Nothing is written to either device. 

I guess I'm throwing in the towel, and finding something else to do with the pi. If you have removal instructions that would be helpful.

dmpop, 2017-05-18:
Sorry it didn't work for you. Delete the *little-backup-box* directory and its cronjob in crontab.

closed 2017-05-18 by dmpop

ElDuderino80, 2017-05-24:
Just an FYI, I got it working again by blowing out the Pi and re-installing the script. I have no idea what got it to the point of not working, but at least it's working as of now. Thanks.