outdoorbits/little-backup-box #12: Not sure what i am doing wrong.

Shanesphotox, 2017-09-23:
Hello,
I'm kind of new to using a raspberry pi. I saw your program and thought it would be a great addition to my workflow,  But somehow i screwed it up setting it up.  I have installed the script and rebooted the pi,  and have connected a drive and the card reader but it does not seem to be working.   Any help with this would be greatly appreciated.

Inrumpo, 2017-09-26:
Same here. I don't get it to work. 
First of all, `curl -sSL https://goo.gl/btmpKd | bash` didn't do anything. The way via terminal did. 

Anyway, the script does not run for me. The Pi does detect my USB stick and my SD card in my cardreader. But that's it. 
Executing `backup.sh` by hand doesn't do anything either. Executing in terminal does nothing as well. 
What I've found: The stick ist not at `.../storage` and the card is not at `.../card`. But changing those in `backup.sh` to the path where my USB stick and my SD card actually are at, doesn't help.

I've run out of ideas for now.

dmpop, 2017-09-27:
> Same here. I don't get it to work. I used `NOOBS 2.4.4` on my Pi 2.
First of all, `curl -sSL https://goo.gl/btmpKd | bash` didn't do anything. The way via terminal did. 

You are supposed to run the command in the terminal.

Try to connect the USB storage device and the card reader. Run then the `sudo lsblk` command. Please post the output here. Also, check the *~/little-backup-box.log* and see whether there are any clues.

Inrumpo, 2017-09-27:
Thanks for your fast response. 

```
pi@raspberrypi:~ $ sudo lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    1 58.9G  0 disk 
└─sda1        8:1    1 58.9G  0 part /media/storage
sdc           8:32   1 29.7G  0 disk 
└─sdc1        8:33   1 29.7G  0 part /media/pi/disk
mmcblk0     179:0    0 14.9G  0 disk 
├─mmcblk0p1 179:1    0  1.6G  0 part 
├─mmcblk0p2 179:2    0    1K  0 part 
├─mmcblk0p5 179:5    0   32M  0 part /media/pi/SETTINGS
├─mmcblk0p6 179:6    0   69M  0 part /boot
└─mmcblk0p7 179:7    0 13.2G  0 part /
```
I can't find a `little-backup-box.log` file in any directory.

st599, 2017-09-27:
It looks like the backup script is expecting /dev/sdb1 to mount on /media/card
See the following lines
```
CARD_DEV="sdb1"
CARD_MOUNT_POINT="/media/card"
```

However, your machine is seeing the card as /dev/sdc1.

Try changing it to

`CARD_DEV="sdc1"`

Inrumpo, 2017-09-27:
Thanks.
By now, I've completely re-installed everything. This time I `used curl -sSL https://goo.gl/btmpKd | bash`.
I did get the `little-backup-box.log` file this time. It's empty. 

```
pi@raspberrypi:~ $ sudo lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sdb           8:16   1 29.7G  0 disk 
└─sdb1        8:17   1 29.7G  0 part /media/pi/disk
sdd           8:48   1 58.9G  0 disk 
└─sdd1        8:49   1 58.9G  0 part /media/pi/Joeys USB
mmcblk0     179:0    0 14.9G  0 disk 
├─mmcblk0p1 179:1    0  1.6G  0 part 
├─mmcblk0p2 179:2    0    1K  0 part 
├─mmcblk0p5 179:5    0   32M  0 part /media/pi/SETTINGS
├─mmcblk0p6 179:6    0   69M  0 part /boot
└─mmcblk0p7 179:7    0 13.2G  0 part /
```
`disk` is my SD card (`32 GB Volume` at `/media/pi/disk`). `Joeys USB` is my USB stick. That looks better than last time. Still no action though.

Shanesphotox, 2017-09-28:
Thank you guys for your help,. I got it to work

jamiek0326, 2017-09-30:
I was having issues too. Couldnt figure out what the story was. I found out that my 'copy-to' drive wasn't formatted correctly which caused the drive identifiers above to be messed up. I switched over to brand new, never touched usb drive and IT WORKED!!! Flawlessly!!!

Im wondering if it is possible for someone who is a little more well versed in the Linux ecosystem to point me in the direction of what format is best for the copy-to drive and how do I properly format it on a mac? In Disk Utility I have ExFAT and MS-DOS (FAT) along with all of the OS X options like APFS, Mac OS, ... as the options. Are either of those FAT options the correct setting?

Im sure, that may be a hangup for those of us on a Mac getting tripped up on this.

dmpop, 2017-10-01:
Thank you all for helping to solve the issue! I added the solution to the [Troubleshooting](https://github.com/dmpop/little-backup-box/wiki/Troubleshooting) wiki page.

closed 2017-10-01 by dmpop