outdoorbits/little-backup-box #22: Combine mode detection depending on attached device

fabiankr, 2018-03-28:
Hi Dmitri,

I've been following your latest changes and the remote control mode looks very promising. I'm planning to build a backup device similar to yours with a Pi Zero but without the USB hub and no WiFi or Bluetooth connection. While on the go I would want to backup to the internal SD card of the Pi and then when I'm home connect an external mSATA drive and backup from the internal SD card to the external storage.

With the remote control mode this is possible, but it requires connecting to the Pi first. It would be really nice if it was possible to do this automatically. In order to detect whether a connected device is a backup source (SD card, camera) or a backup destination (internal SD card, external storage) you could check for the device name, a file similar to the .id file or even the drive size.

For example my camera cards are all <= 128GB but my external drive is 256GB bit. The backup box could then automatically backup from smaller drives than 128GB and backup to external drives which are bigger than 128GB. Of course this doesn't always work but then you could simply add a .backup file to your backup drive and the Pi assumes everything is a backup source unless there's a .backup file on the drive.

Let me know what you think!

dmpop, 2018-03-29:
Thank you for sharing your idea, Fabian! It' certainly doable. In fact, I briefly considered this approach, but ultimately decided against it. This introduces complexity and increases the possibility of errors. I'd prefer to keep things as simple as possible.

dmpop, 2018-03-30:
Sure thing! If you need any help, let me know. Happy coding!

closed 2018-03-30 by dmpop