outdoorbits/little-backup-box #25: Dual Drive Backup

arlojeremy, 2018-04-06:
I'm going travelling in less than a month, and I'd hate to lose the videos and images I'll be taking, but I'm sure it all won't fit on my cameras SD card. Just backing up to one drive is risky, in case something happens to that drive. So, I'd like to plug in 2 portable harddrives, then the sd card, and have it automatically mirror it to BOTH drives.

I have only a minor understanding of linux command line scripting, though I'll take a look myself at modifying it, but I was wondering how hard it would be to make this change.  I'm assuming just mounting another drive and making an additional copy command to that drive as well?

Thank you!

arlojeremy, 2018-04-06:
I was looking at the troubleshooting. Are the drives assigned in order of being input (ie: sda1 is first, sdb1 is second, etc), or is it by USB port (ie: a particular USB port will always be labelled sdb1, etc).

ChemaAjenjo, 2018-04-06:
Hi @arlojeremy (sorry @dmpop for the interruption in this issue)

I have a fork where I implemented a solution for two devices usb (really is a multiple reader card). Maybe this solution can you help you. Only that in this script (`reader-backup.sh`) make a backup into raspberry sdcard and after with another script (`device-backup.sh`) I move this files into other device. If you want you can modify this script `reader-backup.sh` for copy in another device.

Another solution, that can help you if you have internet connection in my fork you have a script (`network-backup.sh`) for update the internal backup to a cloud with `rclone`. But this you need configurate a cloud storage, p.e. I have a owncloud in a raspberry pi, but I'm working for implement with amazon cloud drive.

I hope I've been some help.

dmpop, 2018-04-06:
@arlojeremy I'm not sure whether a setup with two hard disks connected to the Raspberry Pi simultaneously would work. This would draw a lot of power, so you'd need either a decent powered USB hub or a power supply that can deliver enough amperage. Why not just to swap hard disks and run the backup operation twice?

arlojeremy, 2018-04-06:
I have yet to run a test on how long it would take to backup certain amounts of data (for example, a full 64 GB card), so that's my first step, but unfortunately I am still waiting on one of the hard-drives in the mail, and the SD card (and camera) so I can't do a true benchmark and see if the power requirement is too high (I wasn't planning on using a hub, but rather plugging each hard-drive into the RPi3 and then the SD card).

We'll be travelling all day, coming back late and getting up early. If it's possible time-wise to backup to each drive every day, it would be best sleep-wise to have it all run overnight to both drives, instead of having to get up halfway through the night just to swap drives.

Thanks! I'll get back after doing some testing.

closed 2018-05-30 by dmpop