outdoorbits/little-backup-box #27: Not unique media id

abertelmann, 2018-04-21:
Hello.
I came on an issue with identical media ids yesterday when i tried to prepare 3 sd cards in a row.

Preparing for a trip i wanted to prepare the (empty, freshly formated by camera) sd cards and the backup usb stick. I followed the usual way - power up Pi, insert usb backup medium, insert sd card (with usb card reader), wait for shutdown (some 30 seconds). Repeat 3 times in about 3-5 minutes.

When checking the usb stick on another machine i found only 2 backup folders. I checked the sd cards as well and found 2 with identical id file name.

I solved the problem by deleting one card's id file and repeating the setup, about 10 minutes after the first try. I ended OK with a new id there.

I think it may be a problem when preparing more than one media in short time, on a Pi without correct system time (headless Pi zero in my case, no network connection). 

This may lead to loss of data when backing up 2 different cards with identical id onto the same backup medium.

My cards where of 2 different types, the identical ids were on the cards of the same type - any issue there (i did not study the script)?

I just updated little-backup-box to the actual version (as headless backup device) after using the old version for about 8 or 10 months. I did not have any trouble there, maybe because it used random id number or i did not prepare multiple cards in a row. 

Maybe you should revert to the old system, or (since a date-time based system has some advantages for book-keeping) to a combined system. 

As for your work - i really like your little helpers and your ebook - highly recommended!

Greets
Andreas

dmpop, 2018-04-21:
Hi Andreas,

Thank you for reporting the issue! You are right: if the system time is not set, this can cause problems with generating unique ids. I'll work on a fix.

dmpop, 2018-04-21:
Fixed in 162719bd64d6f6430e4d52f6288c7b8598fcf591 I'd be grateful if you could test it.

closed 2018-04-21 by dmpop