outdoorbits/little-backup-box #28: backup box isn't starting

Curtis777, 2018-04-22:
I just installed the image fresh updated on my Raspberry Pi 3B and installed the script sucessfully.

But it didnt start.
The log just says Shutdown scheduled for Mon 2018-04-23 00:46:45 CEST, use 'shutdown -c' to cancel.

i am using an ssd exFAT formated and an SD Card in an Cardreader

Curtis777, 2018-04-23:
Edit: inalso tried an USB Stick in exFAT Format.

Curtis777, 2018-04-23:
got this

root@photobackupPI:/home/pi# cd /home/pi/little-backup-box/scripts/
root@photobackupPI:/home/pi/little-backup-box/scripts# ./card-backup.sh         Shutdown scheduled for Mon 2018-04-23 09:45:15 CEST, use 'shutdown -c' to cancel.
FUSE exfat 1.2.5
WARN: volume was not unmounted cleanly.
fuse: mount failed: Device or resource busy

Curtis777, 2018-04-23:
meanwhile i tried the remote mode as well.
no file has been writen on the usb drive.
is there any thing i forgot to install ?

Curtis777, 2018-04-23:
oh you mean ext3

Curtis777, 2018-04-23:
OK I also tried ext3 and 2 same problem nothing happened

Curtis777, 2018-04-26:
now i tried different usb drives and sd cards and reader...
this is my output:

pi@photobackupPI:~ $ sudo su
root@photobackupPI:/home/pi#
Broadcast message from root@photobackupPI (Thu 2018-04-26 13:07:03 CEST):

Shutdown is activated. To cancel: sudo shutdown -c
The system is going down for poweroff at Thu 2018-04-26 13:11:03 CEST!

Broadcast message from root@photobackupPI (Thu 2018-04-26 13:07:05 CEST):

The system shutdown has been cancelled

root@photobackupPI:/home/pi# ./card-backup.sh
bash: ./card-backup.sh: No such file or directory
root@photobackupPI:/home/pi# cd /home/pi/little-backup-box/scripts/
root@photobackupPI:/home/pi/little-backup-box/scripts# ./card-backup.sh
Shutdown scheduled for Thu 2018-04-26 13:12:48 CEST, use 'shutdown -c' to cancel.
mount: /dev/sda1: can't read superblock
mount: /dev/sdb1 is already mounted or /media/card busy
       /dev/sdb1 is already mounted on /media/card
./card-backup.sh: line 68: [: too many arguments
rsync: link_stat "/media/storage/201804261307-11054.id" failed: No such file or directory (2)
rsync: link_stat "/root/201804261307-18749.id" failed: No such file or directory (2)

DomsProduction, 2018-05-17:
When I insert both USB and SD card, The Pi instantly shuts down, from what I think is supposed to happen is that the files are supposed to transfer than shutdown.

When I turn on the Pi after 2-5 min of letting it stay off while I think of another approach it still has no files transferred.  (It only makes a "CARD_ID")

dmpop, 2018-05-18:
You might want to update Little Backup Box. The version that creates the CARD_ID file is pretty old. Also, try to run the *card-backup.sh* script manually and check whether the output offers any useful clues.

closed 2018-05-30 by dmpop