outdoorbits/little-backup-box #29: camera-backup.sh doesn't copy over all photos taken in burst mode

jealousofyou, 2018-05-13:
If photos have been taken in burst mode then the files can have timestamps with the same hour, minute and second. 

With the current file naming option:
`gphoto2 --get-all-files --skip-existing --filename=%Y%m%d-%H%M%S.%C`
only the first file taken in a particular second will be copied over. The others taken in the same second will be skipped because the name is already taken.

The `--filename` option takes the same time granularity as `date(1)` which only offers nanoseconds as a more accurate timestamp. If you don't want to use that then there might be no option but to include another source of uniqueness (maybe `%f`, the original filename without suffix, e.g. DSC_1234) as part of the target filename on storage medium.

dmpop, 2018-05-24:
Alternatively, you can simple remove the `--filename=%Y%m%d-%H%M%S.%C` part to keep the original file names.

closed 2018-05-24 by dmpop

jealousofyou, 2021-01-27:
Now that the --filename option has been removed from camera-backup.ch, can you also remove the "Rename the transferred files using the YYYYMMDD-HHMMSS.SS format" comment above it?

dmpop, 2021-01-28:
Done. Thanks!

commit c9f031e0 (2018-05-24): Remove --filename parameter

Fix #29