outdoorbits/little-backup-box #31: Little Backup Box Crashing rPi

joshua1187, 2018-05-24:
Hi!

I bought a Raspberry pi b3+ just so I could try out and use little backup box for my photography.

I got the pi booting and running off of a WD Elements external HDD.

since i have installed LBB my pi has started crashing at random times maybe 1-5 minutes. This is without even trying to back anything up.

At one point i was testing its backup capabilities and it looks like it crashed before it was finished downloading the photos.

Here is the LBB log:

Shutdown scheduled for Thu 2018-05-24 16:32:13 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Thu 2018-05-24 16:35:21 UTC, use 'shutdown -c' to cancel.
Saving file as _MG_7959.CR2
Saving file as _MG_7959.JPG
Saving file as _MG_7960.CR2
Saving file as _MG_7960.JPG
Saving file as _MG_7961.CR2
Saving file as _MG_7961.JPG
Saving file as _MG_7962.CR2
Saving file as _MG_7962.JPG
Saving file as _MG_7963.CR2
Saving file as _MG_7963.JPG
Saving file as _MG_7964.CR2
Saving file as _MG_7964.JPG
Saving file as _MG_7965.CR2
Saving file as _MG_7965.JPG
Saving file as _MG_7966.CR2
Saving file as _MG_7966.JPG
Saving file as _MG_7967.CR2
Saving file as _MG_7967.JPG
Saving file as _MG_7968.CR2
Saving file as _MG_7968.JPG
Saving file as _MG_7969.CR2
Saving file as _MG_7969.JPG
Shutdown scheduled for Thu 2018-05-24 16:35:59 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Thu 2018-05-24 21:47:05 UTC, use 'shutdown -c' to cancel.
Skip existing file _MG_7959.CR2
Skip existing file _MG_7959.JPG
Skip existing file _MG_7960.CR2
Skip existing file _MG_7960.JPG
Skip existing file _MG_7961.CR2
Skip existing file _MG_7961.JPG
Skip existing file _MG_7962.CR2
Skip existing file _MG_7962.JPG
Skip existing file _MG_7963.CR2
Skip existing file _MG_7963.JPG
Skip existing file _MG_7964.CR2
Skip existing file _MG_7964.JPG
Skip existing file _MG_7965.CR2
Skip existing file _MG_7965.JPG
Skip existing file _MG_7966.CR2
Skip existing file _MG_7966.JPG
Skip existing file _MG_7967.CR2
Skip existing file _MG_7967.JPG
Skip existing file _MG_7968.CR2
Skip existing file _MG_7968.JPG
Skip existing file _MG_7969.CR2
Skip existing file _MG_7969.JPG
Shutdown scheduled for Thu 2018-05-24 21:50:14 UTC, use 'shutdown -c' to cancel.
Skip existing file _MG_7959.CR2
Skip existing file _MG_7959.JPG
Skip existing file _MG_7960.CR2
Skip existing file _MG_7960.JPG
Skip existing file _MG_7961.CR2
Skip existing file _MG_7961.JPG
Skip existing file _MG_7962.CR2
Skip existing file _MG_7962.JPG
Skip existing file _MG_7963.CR2
Skip existing file _MG_7963.JPG
Skip existing file _MG_7964.CR2
Skip existing file _MG_7964.JPG
Skip existing file _MG_7965.CR2
Skip existing file _MG_7965.JPG
Skip existing file _MG_7966.CR2
Skip existing file _MG_7966.JPG
Skip existing file _MG_7967.CR2
Skip existing file _MG_7967.JPG
Skip existing file _MG_7968.CR2
Skip existing file _MG_7968.JPG
Skip existing file _MG_7969.CR2
Skip existing file _MG_7969.JPG
Shutdown scheduled for Thu 2018-05-24 21:50:35 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Thu 2018-05-24 21:55:51 UTC, use 'shutdown -c' to cancel.

*** Error ***              
An error occurred in the io-library ('Could not claim the USB device'): Could not claim interface 0 (Device or resource busy). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
*** Error (-53: 'Could not claim the USB device') ***       

*** Error ***              
An error occurred in the io-library ('Could not claim the USB device'): Could not claim interface 0 (Device or resource busy). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
*** Error (-53: 'Could not claim the USB device') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --get-all-files --skip-existing

Please make sure there is sufficient quoting around the arguments.

Shutdown scheduled for Thu 2018-05-24 22:00:42 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Thu 2018-05-24 22:00:42 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Thu 2018-05-24 22:05:58 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Thu 2018-05-24 22:05:58 UTC, use 'shutdown -c' to cancel.

*** Error ***              
An error occurred in the io-library ('Could not claim the USB device'): Could not claim interface 0 (Device or resource busy). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
*** Error (-53: 'Could not claim the USB device') ***       

*** Error ***              
An error occurred in the io-library ('Could not claim the USB device'): Could not claim interface 0 (Device or resource busy). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
*** Error (-53: 'Could not claim the USB device') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --get-all-files --skip-existing

Please make sure there is sufficient quoting around the arguments.

Saving file as _MG_7959.CR2
Saving file as _MG_7959.JPG
Saving file as _MG_7960.CR2
Saving file as _MG_7960.JPG
Saving file as _MG_7961.CR2
Saving file as _MG_7961.JPG
Saving file as _MG_7962.CR2
Saving file as _MG_7962.JPG
Saving file as _MG_7963.CR2
Operation cancelled.

Cancelling...
Shutdown scheduled for Thu 2018-05-24 22:06:18 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Thu 2018-05-24 22:06:18 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Thu 2018-05-24 22:18:20 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Thu 2018-05-24 22:18:20 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Thu 2018-05-24 22:23:35 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Thu 2018-05-24 22:23:35 UTC, use 'shutdown -c' to cancel.

FYI i haven't checked but I think the time on my pi is incorrect still. Also, this is all within maybe the last hour as I haven't tested it before now.

Thanks for the help!

joshua1187, 2018-05-27:
Anybody???????

dmpop, 2018-05-27:
If you are using Rapberry Pi with an external hard disk, this might be a power issue. Does your power supply deliver 2.5A? Also, try to run the backup script manually, and when it crashes, run the `dmesg` command. The output may offer some clues.

dmpop, 2018-05-30:
Establish an SSH connection to the Raspberry Pi, then run the following commands:

```
cd little-backup-box
./card-backup.sh
```

closed 2018-06-07 by dmpop

joshua1187, 2018-06-09:
Alright. Sorry its taken me a few days i got busy. When i type ./cd-backup.sh i get "no such file or directory".