outdoorbits/little-backup-box #33: Cannot access '/dev/sd*' no such file or directory

sallan84, 2018-05-30:
Hi, I just set this up and am getting this issue, any help solving would be greatly appreciated! Running on raspbery pi 3 with raspbian installed. 

Thanks, 

Sebastian

dmpop, 2018-05-30:
More info would be nice. What were you doing when you got the error message?

sallan84, 2018-05-30:
Hi, sorry for lack of info, I am mounting and external hard drive and then card reader, getting this error appearing around  10 times in blocks on the log. Also, the shutdown function is tripping the autologin on rasbian, will this affect the script running? Thanks

dmpop, 2018-05-30:
Might be this has something to do with the hard disk's partitioning scheme. Little Backup Box is not intended to work with external hard disks, only with USB sticks. Theoretically, you can make Little Backup Box work with external HDD, but it's tricky to troubleshoot.

sallan84, 2018-05-30:
Ah ok thanks I will try with a USB stick.

closed 2018-05-30 by dmpop

sallan84, 2018-05-30:
Just FYI, i tried with a USB stick and still not working, then I checked what the card reader was mounting as and it is mounting as either SDb or SDe, so it seems that the card reader is the problem. Any suggestions on how to fix this?

sallan84, 2018-05-30:
To clarifiy, it mounts as SDb, then when I unplug and plug in again, it mounts as SDe