outdoorbits/little-backup-box #34: Process complete?

Hofstad, 2018-05-31:
Hi I dont know if this is the correct place to ask. 
I was looking for a script for raspberry like little-backup. But its lacking some features to fully trust it on the go

Is it possible you can add?
- A LED(s) or a smal Oled screen to show the file transfer i % until its done
- Set the Pi in Wifi AP (Access point) for easy control of the device from a android device. For checking what is connected, check the file transfer progress and maybe access the media on the drive.

I dont know if you have time for this but is just a thought

slimline33, 2018-07-29:
Something like this oled would be very nice: https://www.banggood.com/1_3-Inch-4Pin-White-OLED-LCD-Display-12864-IIC-I2C-Interface-Module-For-Arduino-p-1067874.html?p=EX0216559147201405VP

If someone can make  the code I will provide the oled

slimline33, 2018-07-30:
Now I added a buzzer on gpio8 for a short signal when ready.

https://www.raspberrypi-spy.co.uk/berryclip-6-led-add-on-board/berryclip-6-led-add-on-board-instructions/

slimline33, 2018-08-03:
Acutally I'm working on a case for the "CopyPi". A OLED would be nice, but the buzzer is ok. I will use it with a ssd and some sort of 18650 batteries for taking it away 🥇 

[image]

jroblesruiz, 2018-11-17:
@slimline33 I would like to colaborate, I was interested in doing something similar. I have an oled display like that. But It would be great if you share what kind of borad and battery are you using?

JAN4GER, 2019-01-12:
any progress of adding buzzer or oled screen features?

also: how to stop the script while I do not want to use it?
for example when none card in usb, I would like to use Rasbian as usual, but actually your script force reboot after severa seconds.

shunstad, 2019-01-27:
This is more or less the only thing missing from my «pidrop» at the moment, so this would be a welcome feature. While the copy of files usually go without any problems, I would love to get some sort of file verification and an alert of some kind, or even an email with the verfication log, just to be sure that the copy has been completed.

slimline33, 2019-01-27:
Sorry for my late comment. I'm not getting mail notification from github. I will check this tomorrow.

dmpop, 2019-08-30:
The latest version of Little Backup Box introduces experimental support for SSD1306-based 128x32 OLED display.

Austinthemighty, 2020-02-20:
For the oled screen is that over i2c or SPI ?

dmpop, 2020-02-20:
> For the oled screen is that over i2c or SPI ?

I2C

MilzaOnTwitch, 2020-06-26:
I m interested as well and i line to collaborate as well. I wantet to do the same, with display and contactable app.

dmpop, 2020-08-18:
Closing, because I started working on implementing a progress indicator.

closed 2020-08-18 by dmpop