outdoorbits/little-backup-box #38: RPi crashing after little backup box install

joshua1187, 2018-06-16:
Hi, For some reason my issue was closed but not fixed. Heres what i get when i run dmesg in the terminal

pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.34-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1110 SMP Mon Apr 16 15:18:51 BST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000] free_area_init_node: node 0, pgdat 80c84e40, node_mem_map ba3a1000
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:31
[    0.000000] random: fast init done
[    0.000000] percpu: Embedded 17 pages/cpu @ba34a000 s38720 r8192 d22720 u69632
[    0.000000] pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=9c328d51-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 940236K/970752K available (7168K kernel code, 575K rwdata, 2072K rodata, 1024K init, 706K bss, 22324K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c8fd4c   ( 576 kB)
                      .bss : 0x80c96f4c - 0x80d478b4   ( 707 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 25231 entries in 74 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000019] Switching to timer-based delay loop, resolution 52ns
[    0.000269] Console: colour dummy device 80x30
[    0.000288] console [tty1] enabled
[    0.000312] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000327] pid_max: default: 32768 minimum: 301
[    0.000640] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000655] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001587] Disabling memory control group subsystem
[    0.001668] CPU: Testing write buffer coherency: ok
[    0.002059] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002435] Setting up static identity map for 0x100000 - 0x10003c
[    0.002553] Hierarchical SRCU implementation.
[    0.003183] smp: Bringing up secondary CPUs ...
[    0.003869] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.004617] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.005349] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.005453] smp: Brought up 1 node, 4 CPUs
[    0.005463] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.005469] CPU: All CPU(s) started in HYP mode.
[    0.005473] CPU: Virtualization extensions available.
[    0.006326] devtmpfs: initialized
[    0.016993] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.017215] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017233] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.017790] pinctrl core: initialized pinctrl subsystem
[    0.018530] NET: Registered protocol family 16
[    0.021246] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[... 208 lines cut ...]
[   22.233322] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[   22.233878] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 9.10.105 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2018-03-09 18:56:28 
[   25.081409] uart-pl011 3f201000.serial: no DMA platform data
[   26.696678] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   26.696698] brcmfmac: power management disabled
[   27.243853] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   28.018488] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k FS
[   32.542282] Bluetooth: Core ver 2.22
[   32.542481] NET: Registered protocol family 31
[   32.542488] Bluetooth: HCI device and connection manager initialized
[   32.542515] Bluetooth: HCI socket layer initialized
[   32.542529] Bluetooth: L2CAP socket layer initialized
[   32.542561] Bluetooth: SCO socket layer initialized
[   32.594675] Bluetooth: HCI UART driver ver 2.3
[   32.594683] Bluetooth: HCI UART protocol H4 registered
[   32.594685] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   32.594798] Bluetooth: HCI UART protocol Broadcom registered
[   33.070363] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   33.070370] Bluetooth: BNEP filters: protocol multicast
[   33.070385] Bluetooth: BNEP socket layer initialized
[   33.202212] Bluetooth: RFCOMM TTY layer initialized
[   33.202239] Bluetooth: RFCOMM socket layer initialized
[   33.202259] Bluetooth: RFCOMM ver 1.11
[   33.366127] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   35.525095] fuse init (API version 7.26)

joshua1187, 2018-06-16:
Ive just installed a fress copy of rasbian stretch and played some games and other things and have had no crashes. Im going to jow install little backup box and see if the peoblem comes back.

joshua1187, 2018-06-16:
since ive installed a fresh copy of rasbian and little backup box I havn't noticed any crashes. The one thing i did differently was to choose default backup from card rather than from camera.

the new problem is its saying it doesnt have permission to copy the pictures (or somthing like this). Heres the LBB log:

Shutdown scheduled for Sat 2018-06-16 21:52:03 UTC, use 'shutdown -c' to cancel.
rsync: write failed on "/media/storage/201806162251-16069/DCIM/100CANON/IMG_1780.CR2": No space left on device (28)
rsync error: error in file IO (code 11) at receiver.c(393) [receiver=3.1.2]
Shutdown scheduled for Sat 2018-06-16 22:56:37 UTC, use 'shutdown -c' to cancel.
rsync: chown "/media/storage/201806162251-16069/." failed: Operation not permitted (1)
rsync: chown "/media/storage/201806162251-16069/DCIM" failed: Operation not permitted (1)
rsync: chown "/media/storage/201806162251-16069/DCIM/100CANON" failed: Operation not permitted (1)
rsync: chown "/media/storage/201806162251-16069/DCIM/100CANON/IMG_1779.CR2" failed: Operation not permitted (1)
rsync: chown "/media/storage/201806162251-16069/DCIM/100CANON/IMG_1779.JPG" failed: Operation not permitted (1)
rsync: chown "/media/storage/201806162251-16069/DCIM/EOSMISC" failed: Operation not permitted (1)
rsync: chown "/media/storage/201806162251-16069/MISC" failed: Operation not permitted (1)
rsync: write failed on "/media/storage/201806162251-16069/DCIM/100CANON/IMG_1780.CR2": No space left on device (28)
rsync error: error in file IO (code 11) at receiver.c(393) [receiver=3.1.2]
Shutdown scheduled for Sat 2018-06-16 23:00:09 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Sat 2018-06-16 23:06:14 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Sat 2018-06-16 23:12:09 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Sat 2018-06-16 23:12:09 UTC, use 'shutdown -c' to cancel.
mount: /dev/sda1 is already mounted or /media/storage busy
       /dev/sda1 is already mounted on /boot
       /dev/sda1 is already mounted on /media/storage
mount: /dev/sdb1 is already mounted or /media/card busy
       /dev/sdb1 is already mounted on /media/card
rsync: write failed on "/media/storage/201806162251-16069/DCIM/100CANON/IMG_1780.CR2": No space left on device (28)rsync: write failed on "/media/storage/201806162251-16069/DCIM/100CANON/IMG_1780.CR2": No space left on device (28)

rsync error: error in file IO (code 11) at receiver.c(393) [receiver=3.1.2]rsync error: error in file IO (code 11) at receiver.c(393) [receiver=3.1.2]

Shutdown scheduled for Sat 2018-06-16 23:13:46 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Sat 2018-06-16 23:13:46 UTC, use 'shutdown -c' to cancel.
mount: /dev/sda1 is already mounted or /media/storage busy
       /dev/sda1 is already mounted on /boot
       /dev/sda1 is already mounted on /media/storage
Shutdown scheduled for Sat 2018-06-16 23:16:38 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Sat 2018-06-16 23:16:38 UTC, use 'shutdown -c' to cancel.
mount: /dev/sda1 is already mounted or /media/storage busy
       /dev/sda1 is already mounted on /media/storage
mount: /dev/sdb1 is already mounted or /media/card busy
       /dev/sdb1 is already mounted on /boot
       /dev/sdb1 is already mounted on /media/card
rsync: link_stat "/media/storage/201806162311-31514.id" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1196) [sender=3.1.2]
Shutdown scheduled for Sat 2018-06-16 23:17:02 UTC, use 'shutdown -c' to cancel.
Shutdown scheduled for Sat 2018-06-16 23:17:02 UTC, use 'shutdown -c' to cancel.
mount: /dev/sda1 is already mounted or /media/storage busy
       /dev/sda1 is already mounted on /boot
       /dev/sda1 is already mounted on /media/storage
mount: /dev/sdb1 is already mounted or /media/card busy
       /dev/sdb1 is already mounted on /media/pi/EOS_DIGITAL
       /dev/sdb1 is already mounted on /media/card
rsync: chown "/media/storage/201806162251-16069/." failed: Operation not permitted (1)
rsync: chown "/media/storage/201806162251-16069/201806162311-31514" failed: Operation not permitted (1)
rsync: chown "/media/storage/201806162251-16069/201806162311-31514/201806162251-16069" failed: Operation not permitted (1)
rsync: chown "/media/storage/201806162251-16069/201806162311-31514/201806162251-16069/DCIM" failed: Operation not permitted (1)
rsync: chown "/media/storage/201806162251-16069/201806162311-31514/201806162251-16069/DCIM/100CANON" failed: Operation not permitted (1)
rsync: chown "/media/storage/201806162251-16069/." failed: Operation not permitted (1)rsync: chown "/media/storage/201806162251-16069/201806162311-31514/201806162251-16069/DCIM/EOSMISC" failed: Operation not permitted (1)

rsync: chown "/media/storage/201806162251-16069/201806162311-31514/201806162251-16069/MISC" failed: Operation not permitted (1)
rsync: chown "/media/storage/201806162251-16069/201806162311-31514/overlays" failed: Operation not permitted (1)
rsync: chown "/media/storage/201806162251-16069/DCIM" failed: Operation not permitted (1)rsync: chown "/media/storage/201806162251-16069/201806162311-31514" failed: Operation not permitted (1)

rsync: chown "/media/storage/201806162251-16069/DCIM/100CANON" failed: Operation not permitted (1)rsync: chown "/media/storage/201806162251-16069/201806162311-31514/201806162251-16069" failed: Operation not permitted (1)

[... cut ...]
[   32.282311] Bluetooth: HCI UART protocol Broadcom registered
[   32.753357] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   32.753364] Bluetooth: BNEP filters: protocol multicast
[   32.753379] Bluetooth: BNEP socket layer initialized
[   32.796608] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   32.952293] Bluetooth: RFCOMM TTY layer initialized
[   32.952327] Bluetooth: RFCOMM socket layer initialized
[   32.952349] Bluetooth: RFCOMM ver 1.11
[   34.349281] fuse init (API version 7.26)
pi@raspberrypi:~ $ 

Thanks for the help!

joshua1187, 2018-06-17:
update - for some reason it started saving pictures in the BOOT folder.. but i stopped trying to save pictures from the SD card reader and went back to the camera save option (after reinstalling LBB) and it seems to be working correctly now saving pictures in a folder named "BACKUP". Any obvious reason why my system seems to be having so many issues?

thanks again!

joshua1187, 2018-06-18:
Hi, thanks for the reply. I'm running raspbian stretch (not the NOOBs copy if that matters). Recently i reinstalled LBB and it seems to be running fine when downloading from the camera and no more crashing.

Im going to test the process with a different camera and ill try from the SD card reader again to see if its successful.

Ill get back to you with an update hopefully tomorrow.

dmpop, 2018-06-18:
Glad to hear that! Hope it works properly now.

joshua1187, 2018-06-19:
Alright so the Canon 6d worked and the T3i worked however the card reader did not work. does the program default have to be set to read from the card(or does it work similar to a bios boot hierarchy?)? It shuts down the pi but will not pull photos from the card.

Also I read that external HDDs have not been tested so I wanted to let you know that I am using a WD elements 2TB external HDD for my OS and storage and LBB is working fine when pulling images straight from the camera.

dmpop, 2018-06-22:
Are you using the Remote Control mode and trigger the appropriate backup script using the web interface?

Yes, Little Backup Box does work with an external HDD. But this makes the entire setup less portable and more difficult to troubleshoot. That's why I'm not recommending it.

closed 2018-08-10 by dmpop