outdoorbits/little-backup-box #40: Can't copy second card

slimline33, 2018-08-01:
Hi, 

Is it possible to copy a second card? I disabled the shutdown option to copy more than one cards. But the heartbeat stops and the script doesn't recognize the new card.

What can I do?

dmpop, 2018-08-10:
Not sure I understand the problem. Little Backup Box shuts down right after it backed up a card. Just start the device again and back up the second card.

slimline33, 2018-08-21:
At the moment I starting the device again and copy the second card.

There is a 5 min timer in the script before shutdown. I hoped the backup box will check there is a new/another card an it will copy again.

dmpop, 2018-08-23:
The 5-minute shutdown is only in effect if no storage devices are detected. As soon as you plug in a USB storage device, the shutdown is canceled. The Little Backup Box then shuts down immediately after the backup operation is complete. In theory, it should be possible to implement a feature that detects other cards, but that would require serious reworking, for very little gain. If someone is willing to do that, I'll consider merging a pull request, though.

closed 2018-08-23 by dmpop