outdoorbits/little-backup-box #47: Option to delete photos on camera after success

Ryan12439, 2019-07-16:


dmpop, 2019-07-17:
No way. This can potentially be a very dangerous feature. And deleting all the photos using your camera is not that difficult.

closed 2019-07-17 by dmpop

sirtet, 2020-11-21:
@dmpop, how would such an Option be more dangerous than deleting in camera?

I would Think it is Safer if the script does it after a verify.
Data loss would be almost impossible, whereas if I delete on cam, i am not 100% sure if the little box shut down before copying all files, eg. Due to power loss.

Am i wrong?

Anyways, i guess it will be not too hard to add a rm -r to the end of the copy-script...

I too need a fully automatic system, that clears the card before next use. So little box helps a lot, thanks for all the work.

dmpop, 2020-11-22:
It's not about what's more dangerous. It's about using Little Backup Box to keep your data safe. Deleting data on storage cards either in the camera or using Little Backup Box is just asking for trouble. However, if you are an adventurous  type, feel free to add `rm -rf` to the script. :smile:

sirtet, 2020-11-22:
Yes, i understand this is your position.
I like to learn from it.
Would u mind to elaborate where u see trouble arise?
Where/why do u disagree with my thought that a delete by script is safer than manual delete?

sirtet, 2020-11-22:
For my pics to be safe, I need to know with 100% that all files are backed up before i clear the card and take new pics.
How can i manually check this? Open every pic, or maybe see a logfile/message that says backup completed and verified.
A script can do a data integrity check and then delete the source.
If I see the empty card, I know 100% that backup was a success. If pics remain, I know there was an issue.
A shut down pi can not be a good indicator of success, as there are many more ways leading to a shutdown.
That's what I think, and I am thankful if u correct me, before I loose any data.

What is your recommended workflow to check success?

dmpop, 2020-11-24:
Firstly, the backup script uses rsync that performs integrity check during the backup operation. Secondly, even with all the integrity checks in place, there is no guarantee that your data will remain safe (what if you lose the USB stick with your backup data or it turns up to be faulty _after_ you backed up your photos?). Basically, what I'm trying to say is that Linux Backup Box is designed to give you *one more* copy of your file, and not the *only* copy.

sirtet, 2020-11-24:
> what I'm trying to say is that Linux Backup Box is designed to give you one more copy of your file, and not the only copy.

Ok, now that's a clear statement.

So, the following is meant as feedback/help to improve, not to insist on "make me this feature"...

From the readme, i had gotten the impression that it's not only/mainly meant for Backup in the sense of "duplication for security", but more as a tool with which photographers can automate a number of the tasks they regularly have:
- take pics, 
- store them safely, 
- present the pics
- (make room for more pics,) 
- start over...

The readme mentions "Transfer" on bullet point 2 vs. Backup in 1st bullet, this implies to me transfer is different to backup, i interpreted it as "move"...

Now, if you state such a high emphasis on data security (a good thing, i agree), then i think the recommended usage "insert card, wait for shutdown" is dangerous, as described above. A shutdown may have happened due to other reasons than successful rsync, so that's a danger for the user, right?

Maybe that danger should be described in the readme, or fixed... 

- You could replace the shutdown with some led-blinking.
- Or, as i suggested, deleted source-files could be a unmistakable sign for success.
  Sure, the 2nd copy issue is not solved with this. But there could be an option to rsync to more than one destination for that.
- Without deleting the source, another way to communicate success could be to write a jpg back to card, that has text in it "last backup at YMD-H:S successfullly finished.

I just think that many users who like the automatism of little-box, might trust the backup's success and delete their card without double-checking for success.
Because, making room on the card is inevitable at some point, unless you have unlimited cards.
So it would be cool to automate that step too, in a save way. Maybe by default only allow delete  if box can do at least 2 backups to independent destinations.
For the manual delete, i even don't see how i could do a backup-integrity check without repeating what the box does, which would render the whole automatism useless... I think only the remote- option is save, where the UI can give me a success message.

dmpop, 2020-11-24:
Thank you for sharing your thoughts. I'll take your comments under consideration.