outdoorbits/little-backup-box #62: no connection can be established - Wifi / LAN

SchmidL, 2020-08-04:
Hi, after installing the latest version of raspbian lite and the newest from LBB, after the setup, no connection can be establisehd to the RBP 4. 

Wifi doesn't show up and over LAN I also cannot find any IP. 

Any suggestion what to try, where to bugfix?

dmpop, 2020-08-05:
I'm pretty sure that the problem is not related to Little Backup Box. Try to reinstall Raspbian Pi OS without LBB, and if the problem persists, try the Raspberry Pi official forum. Thank you!

SchmidL, 2020-08-05:
hi, well not really, because I install LBB on Raspbian Pi OS via ssh. So I could connect over my LAN. After the installation there was no possibility to connect to the Pi anymore, also no wifi could be found.

What is the specific thing with the SMB / Wins settings? 
Do I have to enable Wifi after the installation separately as I have to do it with the I2C interface?

dmpop, 2020-08-05:
The installer script doesn't touch any network settings at all. Do you use the `curl -sSL https://is.gd/littlebackupbox | bash` command to install LBB?

SchmidL, 2020-08-05:
Yes exactly just this - nothing else.

dmpop, 2020-08-05:
Strange. I reopened the issue, and I'll try to reproduce it. Thanks for your report!

SchmidL, 2020-08-05:
> Strange. I reopened the issue, and I'll try to reproduce it. Thanks for your report!
Before you have to much time invested, just saw that you have a specific script for preparing the SD-card!
There I see you set some important parameters! 

I used this [manual](https://www.raspberrypi.org/documentation/installation/installing-images/mac.md), as I always install Raspian OS that way

dmpop, 2020-08-05:
That's correct: I use the `prepare_card.sh` script to create a bootable microSD card. The script configures Wi-Fi settings. If you use any other way to create a bootable card, you need to configure network settings manually.

closed 2020-08-05 by dmpop