outdoorbits/little-backup-box #63: 3d print case - display support?

ozett, 2020-08-05:
is there any information for the case you showed on the readme.md?
stl-files?
is the display in the case supported by the scripts?

thanks for reply,
ozett

dmpop, 2020-08-05:
What info do you m

> is there any information for the case you showed on the readme.md?
> stl-files?

STL files are included in the repo.

> is the display in the case supported by the scripts?

Yes.

closed 2020-08-05 by dmpop

SchmidL, 2020-08-05:
The stl go provided by @dmpop already in the stl folder. The specific one you're looking for is: https://github.com/dmpop/little-backup-box/blob/master/stl/little-backup-box-rpi4.stl

As you can see is support be given in:
https://github.com/dmpop/little-backup-box/blob/287088c7de5defd28df414e447a333d1f5af2e19/install-little-backup-box.sh#L164

SchmidL, 2020-08-05:
@dmpop does the 128x32 display support still exist as mentioned in  https://github.com/dmpop/little-backup-box/issues/34#issuecomment-526537620

dmpop, 2020-08-05:
> @dmpop does the 128x32 display support still exist as mentioned in [#34 (comment)](https://github.com/dmpop/little-backup-box/issues/34#issuecomment-526537620) ?

No, I remove it.