outdoorbits/little-backup-box #64: enhancement? save smb.conf to .orig with timestamp

ozett, 2020-08-12:
hi, like your project.
as i am testing and run the installscript multiple times on different maschines i thought it might be a good idea to save whatever `smb.conf `exists on the system to multiple .orig with timestamps. so nothing existing/other config gets lost through simple overwriting. what do you think?

https://github.com/dmpop/little-backup-box/blob/master/install-little-backup-box.sh#L110

```
# Configure Samba
#sudo cp /etc/samba/smb.conf /etc/samba/smb.conf.orig
sudo cp /etc/samba/smb.conf /etc/samba/smb.conf.orig-$(date +%Y%m%d%H%M)
```

dmpop, 2020-08-16:
In most situations, you'd probably install Little Backup Box on a fresh system, so there is no need to backup the default Samba configuration. But there is no harm in adding the command, so I'll do it. :-) Thank you for your suggestion!

dmpop, 2020-11-28:
Fixed in https://github.com/dmpop/little-backup-box/commit/81c2695cecfb35b89358cd749e5e2bc1cb355ea5 Sorry it took me so long!

closed 2020-11-28 by dmpop