outdoorbits/little-backup-box #65: Large SD cards not being copied

CradyLab, 2020-08-16:
I was unable to initiate a copy any of my large exFAT cards.  Smaller 16GB cards formatted as FAT32 were no problem, but the 256GB SD cards from my Fuji X-T4 camera and 128GB micro SD cards from my DJI Mavic were ignored and a copy was never initiated.  I tried two different USB 3.0 card readers I but no success (both readers and cards were test-read in Windows 10 with no issues).  This is strange as the pi is able to see and read the contents of the larger cards via the folders on the pi desktop.  The idea behind this project is brilliant, btw.  =)))

As an aside, the external drive used was a USB 3.0 1TB WD My Passport formatted in NTFS - no issues with drive, just the larger cards.

dmpop, 2020-08-16:
I haven't ever tried Little Backup Box with any card larger that 64GB. I don't even have a 128GB SD card handy, so I can't do any testing. It would be great is anyone could try to reproduce the issue.

kschmidt79, 2020-08-17:
Here the same ... 128gb micro sd card does not work. I'm just getting the message that the system will shutdown.
I put in my 32gb sd card and it worked fine. It's not a raspberry, it's a Rock Pi 4 and i'm doing an internal backup on the NVMe SSD. 
But ... when i put the 128gb micro sd card in a sd card adapter and put it in the card reader it works too.

dmpop, 2020-08-17:
Thank you very much for your report! That's really strange. Maybe it's an issue with the exfat-util software. I guess I need to find 126GB and 256GB cards for further testing.

kschmidt79, 2020-08-18:
Thank you for this great software :-) i will also continue to test with different cards and another card reader.

btrunner, 2020-09-26:
Installed LBB on an RPI 4B, mid2020 vintage hardware. When using the latest Raspbian Lite (2020-08-20), LBB appeared to install OK, I2C enabled, but 128x64 OLED display did not work, but I could access WebUI. 

Went back to 2020-02-13 Raspbian Lite image, it installed OK, OLED worked, was able to use 128GB SanDisk USB (SDCZ74-128G-G46, about 1.25 inches long, metal case) as storage for backups. 

Then I did a 3GB backup from USB3 to USB3 and it worked OK. No progress indication in display, (just "working") so need to use WebUI to check progress, especially for larger backups

dmpop, 2020-09-27:
I've just installed Little Backup Box on a fresh copy of Raspberry Pi OS Lite 2020-08-20, it the display works fine. Thank you for reporting back on your results!

> No progress indication in display, (just "working") so need to use WebUI to check progress, especially for larger backups

Yes, the display doesn't show actual progress ( just the _Working..._ message).

thelightdecides, 2020-11-07:
I tried a 128GB SanDisk Extreme Pro SD Card, worked fine, no problems. Card reader was a UGREEN USB 3.0 SD TF Kartenleser USB Card Reader.

dmpop, 2020-11-09:
Thank you for your report!

plzt, 2020-11-30:
I've had no pproblems with a 128GB card on 2 different card readers

closed 2021-07-07 by dmpop