outdoorbits/little-backup-box #67: Unable to select destination folders

plzt, 2020-11-30:
As discussed here: https://discuss.pixls.us/t/custom-backup-directory/21444/8

It is currently not possible to select the destiation folders on the target SD card.

glenerrington, 2022-01-02:
I am using a 256gb usb stick and for me this needs to be addressed fast as the default locations of folders do not map to any location rendering the software useless its a good job I looked before losing any photos

outdoorbits, 2022-01-02:
If I correctly understood your issue, you would like to have an option to configure a base folder at the destination? Same for all destinations?

at USB /basefolder/...
at GoogleDrive /basefolder/...
...

That shouldn't be a big deal. I'm on holiday this week and if you confirm the above you will get this in week 3.

glenerrington, 2022-01-02:
at the moment the defaultt folders are for example /media/storage but my external storage is /media/vault/ so if i can change the folder to /media/vault/storage.

so yes that would be the base folder location

the folders that LBB crate are
/media/storage 
/media/source
/media/cloud
/media/iOS

outdoorbits, 2022-01-02:
Now I think I got you wrong. Lbb mounts as you wrote correctly below /media. But you want to mount it elsewhere? This does not effect the content of your storage. So why?

I thought, you want to effect your storage. If you mount your storage at /media/storage and define another basefolder "vault", my idea was to copy the backup to /media/storage/vault/. So on your device your backup would be below /vault. But this is not what you need?

glenerrington, 2022-01-02:
from what I can tell folders in the /mount directory are on the internal memory and not the external device at /mount/vault which at a later point I can unplug the external storage and access the files on a PC also my sd card is 32Gb which is smaller than my cameras card

Thanks for getting back to me

outdoorbits, 2022-01-02:
External devices are mounted unter /media. Storage for example is mounted at /media/storage. When you disconnect storage and connect it to your PC, root of your device is what you had before at /media/storage.

glenerrington, 2022-01-02:
Sorry but no files are on my external device they are still on the Raspberry Pi.  

To use your example of /media/storeage unless you tell the the OS where the storage should put files it will keep them locally as how else would the system know that you want the files on the external drive or local drive.

external drivers can be mounted in one of two ways 
automatically mounted get put in to /media 

For proper external drives like WD 1 TB Elements Portable External Hard Drive you need to manually edit the fstab and can be mounted anywhere but most tutorials like https://pimylifeup.com/raspberry-pi-mount-usb-drive/ will suggest you cheate a folder in and then mount to /mnt 

As my USB stick is of the automatically mounted the type then any files will need to be placed in the /media/vault example I gave above.

outdoorbits, 2022-01-02:
Do you use an actual version of lbb? If course if you edit fstab, you will kill all the logic of little backup box, this will never work. I don't have any problem with my SanDisk 1TB.
Please just du a new installation, don't change anything (!!!), just use the settings of the GUI. Start the backup by the GUI and follow the instructions of the display or the log you see into the GUI.

glenerrington, 2022-01-02:
I am about to do another fresh install of LBB and I don't need to edit fstab as my USB is a SanDisk Ultra Fit so it's an automount and I did state that my drive would automount

outdoorbits, 2022-01-02:
Automount on Debian (Raspbian) lite?! Or did you install the desktop-version?

glenerrington, 2022-01-02:
full version raspberry pi buster

glenerrington, 2022-01-02:
and again no files transfered to usb drive I am Know projects on here are done by volunteers but I expect projects to work

glenerrington, 2022-01-02:
for proff i am running LBB heres a log output

21:27:33
Backup complete.
Do not unplug!
Power down via
web UI

21:27:30
Backup status:
0 of 0
?%                   

21:27:29
/var/www/little-backup-box/tmp/sync.log: >>>>>
File not found.
<<<<<

21:27:27
Camera o.k.
Working...
Ready
Connect camera

21:27:24
Ready
Connect camera
Ext. storage o.k.
size: 29G

21:27:20
Ext. storage o.k.
size: 29G
free: 25G
Ready

21:27:19
mount_device RESULT=--uuid 54FEC3FAFEC3D282

21:27:17
DEVICE_IDENT_THIS='--uuid 54FEC3FAFEC3D282' ('/dev/sda1') prepared to mount at '/media/storage'

21:27:17
Ready
Insert storage
IP (online):
169.254.167.220

21:27:16
Destination: external 

21:27:16
Source: camera

21:27:01
IP (online):
169.254.167.220
Backup complete.
Do not unplug!

21:26:56
Backup complete.
Do not unplug!
Power down via
web UI

21:26:53
Backup status:
0 of 0
?%                   

21:26:53
/var/www/little-backup-box/tmp/sync.log: >>>>>
File not found.
<<<<<

21:26:50
Camera o.k.
Working...
IP (online):
169.254.167.220

outdoorbits, 2022-01-02:
Did you switch on the camera before connecting? As the log tells, there where zero files transferred.

glenerrington, 2022-01-02:
I have downloaded raspbian_lite-2020-02-14 since the release of bullseye the "legacy" version of the os is only available in full version.

provided that copy of the log to show you I was using LBB

outdoorbits, 2022-01-03:
https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-32-bit
Release date: October 30th 2021

closed 2022-01-13 by outdoorbits