outdoorbits/little-backup-box #73: exFAT not mounting

fakemaxwell, 2021-06-25:
Running into an issue with the exFAT format. I've successfully used LBB with NTFS and FAT32, where it worked normally.  Under System Info I can see that both formats mount correctly.  With exFAT, I see "sda1" show up, but nothing mounts.  Attached a pic of the system info screen.

Is there something else that needs to be done to get exFAT to work correctly?  Thanks.

[image]

dmpop, 2021-06-25:
You have a 1.8TB drive formatted as exFAT? Could you perhaps try a smaller capacity drive?

fakemaxwell, 2021-06-25:
The only small drive I have is 8GB.  exFAT works on it.  Is there a size limitation to exFAT that I'm not aware of?

The idea is to use this to transfer pretty large video files in the field, to be used mostly on Mac computers.  FAT32 has the 4GB limitation and NTFS is read only, although there are workarounds.  I tried using HFS with LBB but it didn't work.

dmpop, 2021-06-28:
> The only small drive I have is 8GB. exFAT works on it. Is there a size limitation to exFAT that I'm not aware of?

HFS is not supported. I might add support for it at some point. But this is not a priority for me. You may want to try to find out hoe to mount ext4-formatted storage devices on Mac.

closed 2021-06-28 by dmpop