outdoorbits/little-backup-box #85: Add function camera to storage and unite all backup-scripts in one

stefans73, 2021-07-11:
Would be nice to have all the four possibilities:

- Backup storage (memory-card) to external storage
- Backup storage (memory-card) to internal storage
- Backup camera to external storage
- Backup camera to internal storage

That's my next target. To reach this without creating another file to maintain, I want to unite all the single backup-scripts in one. This backup.sh will be started with arguments.

dmpop, 2021-07-11:
I'm really interested to see what you come up with. But to be honest, I prefer to have separate scripts for each backup type. They are easier to maintain, and it's easier to add new scripts if necessary. That said, I may change my mind when I see your result. :-)

stefans73, 2021-07-11:
Hi Dimitri,

please just have a look at the pull request following. I had some 
trouble in changes I had do do in every script. Sometimes I did minor 
mistakes in sequence, sometimes I had to rename variables and forgot one 
- nasty things like that. Because I would like to create the 
self-refreshing log-area inside the web-UI, I expect a lot more basic 
changes with a ton of chances for errors.

I'll do some final tests and push it. If you want to test it, please run 
change-mode.sh to get the correct crontab-entry. Everything should also 
run via web-UI.

Thank you,
Stefan

---
Mit meinem angefügten öffentlichen Schlüssel "0xE61B756E.asc" können Sie 
gerne verschlüsselte Mails an mich senden. Näheres auch unter 
https://www.gpg4win.de/

Links:
------
[1] 
https://github.com/dmpop/little-backup-box/issues/85#issuecomment-877809559
[2] 
https://github.com/

closed 2021-07-13 by stefans73