outdoorbits/little-backup-box #96: IP not displayed correctly after installing comitup

ruebyi, 2021-07-20:
Hi!

I love your project and got it running very quickly.

But One bug is still there - or I did a mistake:
After installing comitup I won't get the display to show the actual IP. It sticks with 10.42.0.1

Before installing comitup it worked!

Best Tobias

dmpop, 2021-07-21:
Thanks for reporting the bug. I'll look into it.

ruebyi, 2021-07-23:
I just played around a bit - I receive the correct Adress via the Mail Notification, but OLED and the Logmonitor show the comitup Adress

dmpop, 2021-07-23:
Good to know! Thank you for reporting back. :-)

ruebyi, 2021-08-01:
With the latest fixes it‘s better - it shows the Actual IP on top 

192.168.1.116
LittleBackupBox
10.42.0.1
Ready

dmpop, 2021-08-01:
Good to know. Thank you for reporting back!

closed 2021-08-01 by dmpop