outdoorbits/little-backup-box #99: OLED Permission denied

maelremrem, 2021-07-22:
I have a problem with my oled screen
The OLED won't start but with the i2cdetect -y 1 I can see it (only if I repeat 3 times the command)
It blinks and some of the text appears when I run the status-display.sh
This is the output on the ssh window
[image]

any idea to fix this ?

stefans73, 2021-07-23:
Did you run status-display.sh as root?

---
Mit meinem angefügten öffentlichen Schlüssel "0xE61B756E.asc" können Sie 
gerne verschlüsselte Mails an mich senden. Näheres auch unter 
https://www.gpg4win.de/

Links:
------
[1] 
https://user-images.githubusercontent.com/11769158/126715547-2718fc66-3c8e-42d1-b443-d8d595f06779.png
[2] https://github.com/dmpop/little-backup-box/issues/99
[3] 
https://github.com/

maelremrem, 2021-07-23:
Yes and it's blocked to this :
[image]
and if I try dos2unix manually, it works to convert but this is what I get : 
[image]

maelremrem, 2021-07-23:
I made a copy of the status-display.sh script and modified it to comment this : "dos2unix "$CONFIG"
and now this is what I get on the OLED : https://youtube.com/shorts/I-iMOUvJ20Q?feature=share

dmpop, 2021-07-23:
Maybe it's a good idea to start with some basic troubleshooting first.

Try to run the following commands on the Raspberry Pi:

```
sudo oled r
sudo oled +a "Hello"
sudo oled +b "this"
sudo oled +c "is a"
sudo oled +d "test"
sudo oled s
```

Does it work? Do you see the text on the screen?

maelremrem, 2021-07-23:
it does nothing. No text on screen and no error on the console
when I do sudo i2cdetect -y 1 this is what I get :
[image]
I have to do it 2 times to get an address
I use an RPI3 B+ does it change something ?

dmpop, 2021-07-23:
If the display shows nothing, and the Raspberry Pi has trouble detecting it (you need to run the detect command twice), it either mean that the display is defective or it's wired incorrectly to the GPIO pins.

closed 2021-07-23 by dmpop