outdoorbits/little-backup-box #100: Enhancment: More Links in Mail?

ruebyi, 2021-07-24:
Hi,

I'm quiet sure it's just a small command, but I'm not able to add it my self... 

I just thought it would be great to get not only one Link to the WebUI in the HTML Mail, but also the Link for Upload, FTP and Fileserver - it would be much faster just to click those links than adding the Port manually.

dmpop, 2021-07-24:
Good idea! I'll add it to my list. :-)

ruebyi, 2021-07-24:
After Trying Mejiro - perhaps it would be possible to choose the needed adresses in the settings? Or put them there yourself?

dmpop, 2021-07-30:
Stefan has already added the **Files** link to the wb UI, and I think he's already working on including Mejiro.

stefans73, 2021-07-30:
Hello,

yes, I hope to find time to test my changes later and push them tonight.

Stefan

stefans73, 2021-07-31:
done

closed 2021-07-31 by stefans73