outdoorbits/little-backup-box #113: Progress display

thelightdecides, 2021-07-30:
Hello,

after installing the latest version I have a problem with the progress on the display and the log monitor in the web interface. The progress stops at about 90-100% and nothing happens after that. The Pi does not shut down. When I look at the transferred files, everything is backed up from the sd card to the ssd.

Best regards
Thomas

dmpop, 2021-07-30:
I fixed this one two days ago. I've just tried it and the fix works.

thelightdecides, 2021-07-30:
I tried yesterday...I will try again later today.

thelightdecides, 2021-07-30:
I tried again, completely new installation, same issue....

dmpop, 2021-07-30:
Thanks for reporting back! Two questions:

1. What type of backup do you run (source external or something else)?
2. Do you run the backup automatically on boot (via cronjob), or do you use web UI?

thelightdecides, 2021-07-30:
1. Source external (SD-Card -> SSD)
2. automatically

dmpop, 2021-07-30:
Ah, I think I've found the problem. Looks like Stefan broke my fix. :-) Could you try this:

On the Raspberry Pi, open the `little-backup-box/scripts/backup.sh` file for editing. Replace this line 380

```
source "${WORKING_DIR}/poweroff.sh poweroff"
```

with this:

```
source "${WORKING_DIR}/poweroff.sh"
```

thelightdecides, 2021-07-30:
Yes, this was the problem. With the modified line, it works fine!

dmpop, 2021-07-30:
Excellent! I've reverted the "unfix" :-) Thank you for following up on this!

closed 2021-07-30 by dmpop

stefans73, 2021-07-30:
Hi,

I do not understand, why this happend. But because poweroff.sh has to 
handle both, poweroff and reboot, it was intended to use this switch 
obligate. I will check that, but index.php also calls "poweroff.php 
poweroff" or "poweroff.php reboot" and it works. It's just working, 
because at the end the script just asks for it reboot else... And "else" 
is the winner. ;-)

Stefan

---
Mit meinem angefügten öffentlichen Schlüssel "0xE61B756E.asc" können Sie 
gerne verschlüsselte Mails an mich senden. Näheres auch unter 
https://www.gpg4win.de/

Links:
------
[1] 
https://github.com/dmpop/little-backup-box/issues/113#issuecomment-889892806
[2] 
https://github.com/

dmpop, 2021-07-30:
Yeah, it was unclear to me too why `source "${WORKING_DIR}/poweroff.sh poweroff"` wasn't working. That's why I solved the problem with the `if ... poweroff else reboot` tweak. The important thing is it works now. :-)

stefans73, 2021-07-30:
Hi, did you set POWER_OFF=true in config.cfg?

---
Mit meinem angefügten öffentlichen Schlüssel "0xE61B756E.asc" können Sie 
gerne verschlüsselte Mails an mich senden. Näheres auch unter 
https://www.gpg4win.de/

Links:
------
[1] 
https://github.com/dmpop/little-backup-box/issues/113#issuecomment-889892806
[2] 
https://github.com/

dmpop, 2021-07-30:
Yes, of course. :-)