outdoorbits/little-backup-box #115: One SD Card won

ruebyi, 2021-07-31:


dmpop, 2021-07-31:
No idea what causes the problem. Perhaps something to do with permissions? Stefan has fixed a few problems related to `sudo` in the backup script. Can you try the latest version?

ruebyi, 2021-08-01:
Seems to work, but let me verify in hohe next days

dmpop, 2021-08-01:
Excellent! Feel free to close the issue if the solution works for you. 😃

ruebyi, 2021-08-07:
So it happened again.

First attempted worked perfect. Two days later - a Card that already worked can‘t be backuped anymore…

ruebyi, 2021-08-07:
2021/08/01 23:01:39 [4851] rsync: chown "/media/storage/202108012301-31863/." failed: Operation not permitted (1)
2021/08/01 23:01:39 [4851] .d...pog... ./
2021/08/01 23:01:39 [4851] Number of files: 1 (dir: 1)
2021/08/01 23:01:39 [4851] Number of created files: 0
2021/08/01 23:01:39 [4851] Number of deleted files: 0
2021/08/01 23:01:39 [4851] Number of regular files transferred: 0
2021/08/01 23:01:39 [4851] Total file size: 0 bytes
2021/08/01 23:01:39 [4851] Total transferred file size: 0 bytes
2021/08/01 23:01:39 [4851] Literal data: 0 bytes
2021/08/01 23:01:39 [4851] Matched data: 0 bytes
2021/08/01 23:01:39 [4851] File list size: 0
2021/08/01 23:01:39 [4851] File list generation time: 0.001 seconds
2021/08/01 23:01:39 [4851] File list transfer time: 0.000 seconds
2021/08/01 23:01:39 [4851] Total bytes sent: 52
2021/08/01 23:01:39 [4851] Total bytes received: 110
2021/08/01 23:01:39 [4851] sent 52 bytes  received 110 bytes  324.00 bytes/sec
2021/08/01 23:01:39 [4851] total size is 0  speedup is 0.00
2021/08/01 23:01:39 [4851] rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
2021/08/01 23:02:39 [1273] building file list
2021/08/01 23:02:40 [1273] rsync: chown "/media/storage/202108012301-31863/." failed: Operation not permitted (1)
2021/08/01 23:02:40 [1273] .d...pog... ./
2021/08/01 23:02:40 [1273] Number of files: 1 (dir: 1)
2021/08/01 23:02:40 [1273] Number of created files: 0
2021/08/01 23:02:40 [1273] Number of deleted files: 0
2021/08/01 23:02:40 [1273] Number of regular files transferred: 0
2021/08/01 23:02:40 [1273] Total file size: 0 bytes
2021/08/01 23:02:40 [1273] Total transferred file size: 0 bytes
2021/08/01 23:02:40 [1273] Literal data: 0 bytes
2021/08/01 23:02:40 [1273] Matched data: 0 bytes
2021/08/01 23:02:40 [1273] File list size: 0
2021/08/01 23:02:40 [1273] File list generation time: 0.001 seconds
2021/08/01 23:02:40 [1273] File list transfer time: 0.000 seconds
2021/08/01 23:02:40 [1273] Total bytes sent: 56
2021/08/01 23:02:40 [1273] Total bytes received: 110
2021/08/01 23:02:40 [1273] sent 56 bytes  received 110 bytes  110.67 bytes/sec
2021/08/01 23:02:40 [1273] total size is 0  speedup is 0.00
2021/08/01 23:02:40 [1273] rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
2021/08/01 23:05:14 [1399] building file list
2021/08/01 23:05:14 [1399] rsync: chown "/media/storage/202108012301-31863/." failed: Operation not permitted (1)
2021/08/01 23:05:14 [1399] .d...pog... ./
2021/08/01 23:05:14 [1399] Number of files: 1 (dir: 1)
2021/08/01 23:05:14 [1399] Number of created files: 0
2021/08/01 23:05:14 [1399] Number of deleted files: 0
2021/08/01 23:05:14 [1399] Number of regular files transferred: 0
2021/08/01 23:05:14 [1399] Total file size: 0 bytes
2021/08/01 23:05:14 [1399] Total transferred file size: 0 bytes
2021/08/01 23:05:14 [1399] Literal data: 0 bytes
2021/08/01 23:05:14 [1399] Matched data: 0 bytes
2021/08/01 23:05:14 [1399] File list size: 0
2021/08/01 23:05:14 [1399] File list generation time: 0.001 seconds
2021/08/01 23:05:14 [1399] File list transfer time: 0.000 seconds
2021/08/01 23:05:14 [1399] Total bytes sent: 52
2021/08/01 23:05:14 [1399] Total bytes received: 110
2021/08/01 23:05:14 [1399] sent 52 bytes  received 110 bytes  324.00 bytes/sec
2021/08/01 23:05:14 [1399] total size is 0  speedup is 0.00
2021/08/01 23:05:14 [1399] rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]

dmpop, 2021-08-07:
Really strange. It's the same card you had problems the first time?

ruebyi, 2021-08-07:
Nope, another card

stefans73, 2021-08-07:
Hi,

could you please send the content of Devices and of Disk space in System info after broken sync?

Thank you,
Stefan

ruebyi, 2021-08-07:
Devices
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 232.9G  0 disk 
└─sda1        8:1    0 232.9G  0 part /media/storage
sdb           8:16   1 116.6G  0 disk 
└─sdb1        8:17   1 116.6G  0 part 
mmcblk0     179:0    0 119.4G  0 disk 
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0 119.1G  0 part /
Disk space
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       126G  2.3G  119G   2% /
devtmpfs        1.8G     0  1.8G   0% /dev
tmpfs           2.1G     0  2.1G   0% /dev/shm
tmpfs           2.1G   11M  2.1G   1% /run
tmpfs           5.3M  4.1k  5.3M   1% /run/lock
tmpfs           2.1G     0  2.1G   0% /sys/fs/cgroup
/dev/mmcblk0p1  265M   52M  213M  20% /boot
/dev/sda1       251G  112G  139G  45% /media/storage

stefans73, 2021-08-07:
That means, sdb1 is not mounted, but should be mounted as source, right? Do you use an active USB hub? Maybe it could be a power problem.
I have little bit different problems with xqd-cards and I'm working on a more robust mount-strategy. I will tell you ...

Stefan

ruebyi, 2021-08-07:
Sdb1 is a Gibot Cardreader. 
it worked already with that card…

stefans73, 2021-08-07:
But it isn't mounted. Could you check it with an active hub?

ruebyi, 2021-08-07:
I will test it at home - I‘m travelling back tomorrow.

there I will be able to test another Cardreader as well

ruebyi, 2021-08-08:
OK, at home:

Raspi with original Power Supply - Same error - no Card is recognized correctly
Raspi with another SD Card Reader - SD Card Reader is not Compatipble (Transcend UHSII Reader)
Raspi with Active Hub - Same Error - no Card is recognized Correctly
Raspi with Active Hub - SD Card Reader and external SSD connected to Hub - No Card is recognized correctly

Even Cards, that already worked are not used correctly anymore.

after all this attempts WebUI was not responisive anymore.

ruebyi, 2021-08-16:
Hi, 
I just tested the new Build:

Device error!

Type: storage to external
Source ID: 202108012223-32238

Backup log:
...

Number of files: 1,835 (reg: 1,826, dir: 9)
Number of created files: 1,834 (reg: 1,826, dir: 8)
Number of deleted files: 0
Number of regular files transferred: 1,826
Total file size: 29.45G bytes
Total transferred file size: 29.45G bytes
Literal data: 29.45G bytes
Matched data: 0 bytes
File list size: 0
File list generation time: 0.001 seconds
File list transfer time: 0.000 seconds
Total bytes sent: 29.46G
Total bytes received: 34.76K

sent 29.46G bytes  received 34.76K bytes  29.53M bytes/sec
total size is 29.45G  speedup is 1.00

I'm not able to see what the error is...

dmpop, 2021-08-16:
I had the same error. I reformatted the backup USB storage device, and the error was gone. Please note that both developers are traveling now, so we won't be able to do any troubleshooting in the next couple of weeks.

ruebyi, 2021-08-16:
Have a good journey! 
I will try to format and report back

dmpop, 2021-08-19:
I've identified one possible cause of the "Lost device" error. It has something to do with the USB 3 ports of Raspberry Pi. If two devices connected to both USB 3 ports draw too much power, the Raspberry Pi "loses" one of them. It seems to be some sort of hardware limitation, and the only way to prevent this from happening is to use one USB 2 and one USB 3 port. Of course, this means a slower transfer speeds, but there is nothing can be done about it, I'm afraid.

ruebyi, 2021-08-19:
Ok, Strange That it worked for some Cards then… 
Thank you!

stefans73, 2021-08-19:
What happens, if you try to mount the device in webUI/tools?

thelightdecides, 2021-08-23:
I have a similar problem. When using CFexpress cards, the copy process stops after about 1500 files. Either the SSD or the memory card is no longer found. Using an active hub or using a USB 2 and a USB 3 port solves the problem. 
The problem does not occur with SD cards, even 256GB SD cards are copied without problems.

dmpop, 2021-08-23:
I have a suspicion that it might have something to do with overheating. It happened to me a couple of times too when I was transferring 1400+ files. I noticed that the USB stick and the USB ports on Raspberry PI got very, very hot. I guess next time I do a backup, I'll stick my Raspberry Pi into a fridge and see it it helps. 😄

dmpop, 2021-09-06:
I've rolled back the code in the `main` branch to the last stable version. So this bug is no longer valid for the current version of Little Backup Box.

The newest code with all the improvements (and bugs :wink: ) is now in the `development` branch. I've assigned this bug to Stefan. Hopefully he'll be willing to fix it.

stefans73, 2021-09-06:
Hi,

of course, but I'm still in holiday. Next week I'll be back and try to bugfix.

Have a good time!
Stefan

stefans73, 2021-09-13:
I see two issues:
1. "Operation not permitted" could mean the card is corrupted. Please check by the fsck-tool.
2. The message "Device error!" means, the device is not mounted anymore directly after the backup. Probably this is not a software-issue but a hardware-issue. Could you please check in "System info": Is the device is still mounted? If yes, everything worked fine and there would be a problem about the error-detection-routine that pushes "Device error!". If not, I would expect a hardware-problem. If the device was lost, it would be possible to code a loop witch retries the backup for a maximum of config-defined times. If it's a problem of overheating as Dimitri mentioned, a second run after cooling down could be the solution. In a second run rsync just checks date and size of existing files, so it is much faster and missing files could be backuped.

ruebyi, 2021-11-20:
Hi, is there something I can do to support you on this issue?

stefans73, 2021-11-20:
Hi,

I'm just hunting for a bug, then I would like to contact you for testing again.

Best,
Stefan

stefans73, 2021-11-22:
Hi,

again, I did many changes and bugfixes and my last tests have been 
successfully. All my last contribs are not yet in 
https://github.com/dmpop/little-backup-box but in my 
https://github.com/stefans73/little-backup-box . I am very excited to 
get (hopefully good!) news from you. Many technical details are 
fundamental different to the main branch now, so installation on a new 
bullseye-lite is necessary.

Thank you for testing!
Stefan

---
Mit meinem angefügten öffentlichen Schlüssel "0xE61B756E.asc" können Sie 
gerne verschlüsselte Mails an mich senden. Näheres auch unter 
https://www.gpg4win.de/

Links:
------
[1] 
https://github.com/dmpop/little-backup-box/issues/115#issuecomment-974628098
[2] 
https://github.com/

stefans73, 2021-11-22:
Hi Dima,

last mail got wrong, I wanted to ask ruebyi to test. But I would be very 
happy to get some feedback from you, if you find the time. I spent a 
night with problems about filebrowser and ended up with replacing it by 
tinifilemanager. This is/was my trouble: 
https://stackoverflow.com/questions/70050961/mount-by-php-script-apache 
but now it works great!

Thank you,
Stefan

---
Mit meinem angefügten öffentlichen Schlüssel "0xE61B756E.asc" können Sie 
gerne verschlüsselte Mails an mich senden. Näheres auch unter 
https://www.gpg4win.de/

Links:
------
[1] 
https://github.com/dmpop/little-backup-box/issues/115#issuecomment-974628098
[2] 
https://github.com/

ruebyi, 2021-11-24:
I will test your build tonight I think!

ruebyi, 2021-11-24:
Seems to work great! 
I like the new Features and the new Filebrowser

stefans73, 2021-11-25:
Hi,

I'm really relieved to read this. Did you also test the cards causing problems before?
Could you please tell me, which camera you use? Did you backup by cardreader or directly from camera? I would like to publish compatible systems.

Thank you,
Stefan

ruebyi, 2021-11-25:
OK, 
Yesterday I used 2 UHS II SD Cards from Olympus Cameras - If you want I can test today Micor SDs from DJI and GoPro along with Olympus Cards

I'm using a cheap Cardreader from Gibot.

I will report when I Backkuped the next four to five Cards that I have here with me

stefans73, 2021-11-25:
Hi,

of course, please let me know about. I will add all this to 
https://github.com/stefans73/little-backup-box/wiki/Compatibility

Thank you!
Stefan

thelightdecides, 2021-11-25:
Hi,
thank you very much for trying to solve the problem. I installed your version and made a short test. Unfortunately, the backup did not complete. During the first test I got this error message: "rsync: [sender] Write error: broken pipe (32)". After rebooting, I tried again, after a few hundred files: "2021/11/25 22:07:55 [4317] rsync: [sender] send_files could not open "/media/source/fill/BJ6A6757.CR3": Input/Output error (5) ".
Devices
NAME MAJ:MIN RM SIZE RO TYPE MOUNTPOINT
sda 8:0 0 931.5G 0 disk 
`-sda1 8:1 0 931.5G 0 part /media/storage
sdc 8:32 0 119.2G 0 disk 
`-sdc1 8:33 0 119.2G 0 part 
mmcblk0 179:0 0 29.8G 0 disk 
|-mmcblk0p1 179:1 0 256M 0 part /boot
`-mmcblk0p2 179:2 0 29.6G 0 part /
Disk space
File system Size Used Available Usage% Mounted in
/dev/root 32G 2.3G 28G 8% /
devtmpfs 1.9G 0 1.9G 0% /dev
tmpfs 2.1G 0 2.1G 0% /dev/shm
tmpfs 806M 2.2M 803M 1% /run
tmpfs 5.3M 4.1k 5.3M 1% /run/lock
/dev/mmcblk0p1 265M 51M 214M 20% /boot
/dev/sda1 1.1T 143G 858G 15% /media/storage
/dev/sdb1 129G 128G 39M 100% /media/source

I think the Pi has lost the reader again, and it was not remounted correctly.
Unfortunately, I don't know much about Linux and can't help finding the problem.

Hardware I used: Samsung T5 SSD, SanDisk Extreme Pro CFexpress 128GB, Delkin Card-Reader.

I will try with another cardreader tomorrow and also with SD cards.

Best regards
Thomas

stefans73, 2021-11-26:
Hi,

for me there are a lot of questions open. I will upload a new version next week with extended logging options. If you would spend some time for testing again and send me that logfile, I hope the problem will get clear.

Thank you,
Stefan

ruebyi, 2021-11-26:
Today - same Hardware - a second card didn't work.
The same Card and the same Reader like two days ago...

stefans73, 2021-11-28:
A new version is online (https://github.com/stefans73/little-backup-box). I changed a lot about logging and also a few things about restarting the backup in case of an error. Please help me to address the problem and send me your log file. Please have a look at https://github.com/stefans73/little-backup-box/wiki/Debuggin-and-log-file first.

Thank you,
Stefan

thelightdecides, 2021-11-29:
I again used a CFexpress card and the original Pi power supply. The first backup runs fine, no error. Then I deleted the .id file on the card and tried again. Now the card should be a new card for the backup, but its looks like the LBB just wanted to copy 188 files? And the device is lost. I will try again later.

[little-backup-box-128GB-CFexpress-Pi-Netzteil-2-Error.txt](https://github.com/dmpop/little-backup-box/files/7619194/little-backup-box-128GB-CFexpress-Pi-Netzteil-2-Error.txt)

stefans73, 2021-11-29:
The problems seems to be on the destination-device...

thelightdecides, 2021-11-29:
Yes, i connected the SSD to my Windows computer and it shows an error. I found 2587 files has been copied before the pi lost the device.

Holen Sie sich Outlook für Android<https://aka.ms/AAb9ysg>

thelightdecides, 2021-11-29:
Here is one more log-file:
[little-backup-box (3).log](https://github.com/dmpop/little-backup-box/files/7621059/little-backup-box.3.log)

stefans73, 2021-11-29:
Here both devices got lost (maybe at the same time) and they are even 
not visible for linux lsblk anymore. They are like 
hardware-disconnected! Strange.

But this seems to happen after long intensive activity only? Loosing 
both devices after a long period of high activity - could it be, the 
controller of the pi is overheated? What if you power down the rpi after 
a broken backup, wait some time to cool down and retry? (Maybe you have 
to repair the partitions first, use webUI/Tools.) Only files not synced 
before are planed to sync and synced in secondary tries. So you maybe 
can backup step by step.

To get perhaps a little more information about mount-problems, I 
modified the log a little (catch message of mount-command). If you find 
the time, could you please:

  	* install the latest version
  	* choose Default backup mode: none
  	* set the log-level to 3 (and activate sync-log)
  	* delete log
  	* connect the devices
  	* Tools: Check and repair both relevant partitions
  	* Tools: If partitions are mounted: click Remove USB 1/ Remove USB 2
  	* unplug the source device
  	* Start backup and connect the source device when asked for
  	* download the log file ...

Maybe easier: Can you copy all the files in one session from your source 
to your ssd using your windows-machine? If not, it could be a 
device-specific problem.

thelightdecides, 2021-11-29:
ok, I try do this tomorrow.

Probably not the best solution, but could you automatically stop the backup process after e.g. 1000 files and restart it, maybe with a short pause (maybe the temperature on some components rises too far and then the system becomes unstable). The problems usually occur after 1500 or more files. Then you could probably also prevent the errors on the drives. 
The problems occur only with the CFexpress cards, SD cards of the same size and also larger are copied without problems. I used a 256GB SD card and the backup was no problem.

stefans73, 2021-11-30:
Breaking the process by timeout also means loosing information about what's synced. I did anther approach. If a device-lost occurs and the sync-process took longer than an defined time and returned an error-code, it expects overheating and waits some time before next retry for cooling.
For now I suspect overheating, if sync took >= 120 sec. In this case there is a break of 30 sec for cooling. Unfortunately I can't test this. Could you tell me, after minimal how many seconds this error occurs? Are 30 sec cooling sufficient?
You find the updated version at https://github.com/stefans73/little-backup-box

thelightdecides, 2021-12-01:
I haven't had time to test this yet. I hope tomorrow.

I have tried another SSD. SanDisk Extreme portable 2TB in combination with the CFexpress card reader. Then the system becomes even more unstable. With earlier versions of LBB I once used an active hub. When I try that now, the drives are mounted but the copy process either doesn't start at all or it is extremely slow. Have you tried this? Can you confirm that an active hub still works? I believe some devices require more or more stable power to operate. My very cheap USB power meter shows very fluctuating values on the Pi at the SSD and card reader. On the laptop this is much more stable. In the end, the whole thing needs to work later with a powerbank if you want to backup pictures on the go. Maybe an active hub solves all the problems.

stefans73, 2021-12-01:
I´m using XQD-cards and I had a lot of trouble (only) with them and I 
guess, you have the same. As you described, I also had unstable power 
situations. Since about one week I use a powerd USB-hub and everything 
is working stable and without any errors! Yes, it works with an active 
hub! I use an Anker USB 3.0 70W device, but of course, that´s not what I 
need for travelling. If you know an active hub usable with a powerbank, 
I´m very interested! On the other side, XQD-backup to my Sandisk 
portable extreme 1TB SSD was more stable in combination with my 
powerbank than an 5V/3A power supply.

When you find the time, I would be very interested in these times:

1. backup start to break
2. break until devices are visible again (in System Info)

thelightdecides, 2021-12-02:
Some time ago I used this with a HDD. https://www.amazon.de/CSL-Verteiler-Ultrabook-Tablet-PC-bus-powered/dp/B01K7RR3W8/ref=nav_ya_signin?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&keywords=usb+hub+csl&qid=1638484908&sr=8-6& .That had worked well. Instead of the power supply, I connected the hub to my powerbank (https://www.thelightdecides.com/reise-fotografie-blog/little-backup-box) . But at the moment the hub makes problems in combination with the SSD and/or the card reader. 
I have tried the new program. On two runs it worked with the 30s, the card reader was mounted again and the copy process completed. On one run, both devices were lost and could not be reconnected, even not manually via the mount button. I don't know if it is possible to use less than 30s (I was too late to watch the remount), if it is a temperature problem, the time depends on the ambient temperature.

[little-backup-box-fehler-kein-mount-moeglich.txt](https://github.com/dmpop/little-backup-box/files/7645619/little-backup-box-fehler-kein-mount-moeglich.txt)
[little-backup-box-source-remount.txt](https://github.com/dmpop/little-backup-box/files/7645626/little-backup-box-source-remount.txt)

stefans73, 2021-12-03:
Thank you for the link to the powered hub. Maybe it's worth to test it for me.
In your log of the failed remount the source device does not appear again at all. 30 seconds for cooling may be too short, but I don't know if Debian will automatically detect the device after that failure again anyway. Could your try to unplug the source device in case of that error and reconnect?
I'm going to extend the time for cooling, do you think 60 seconds could be better?

stefans73, 2021-12-03:
I just visited your website https://www.thelightdecides.com/ . Great images, I'm deeply impressed!
The time for cooling I set to 60s now. Then I read on your website, you would like to know the estimated time to finish. Great idea. Done.

thelightdecides, 2021-12-03:
Many thanks!!!

I will try again with the new version. I also have 2 other hubs that I will test, both have a USB-C input for the power supply, which I like much better than the micro USB port of the other hub. I won't have time this weekend, but I'll get back to you as soon as I can.

stefans73, 2021-12-03:
Could you please tell me (and us) which hubs these are? The micro USB indeed was the reason I didn't order yet. So tests with these hubs also would be extremely interesting.

thelightdecides, 2021-12-03:
ORICO 4-Port USB3.0 HUB (this has a short cable, better for traveling)
https://www.amazon.de/gp/product/B08HVGCH8B/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

UGREEN 4-Port USB 3.0 HUB
https://www.amazon.de/gp/product/B08KXNDTPV/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

but I didn't test them....

stefans73, 2021-12-04:
Thank you. I just ordered the ORICO and hope this will my solution for travelling. This morning I put ~50GB images on a Sony XQD. As storage davice I used my SanDisk Extreme Portable SSD 1 TB. This is what I got:

- SSD @ hub; QXD @ hub: No problem
- SSD @ pi; QXD @ hub: No problem
- XQD @ pi; SSD @ pi: backup impossible
- XQD @ pi; SSD @ hub: backup impossible

In the last cases the XQD-reader seemed to fall into an error-condition and was not listed anymore. After disconnect and reconnect it appeared again but was lost after seconds. The display turned to black. While the XQD was connected directly to the pi, I had 4.7 V and less on the USB-ports. I think the problems described before, are more or less related to power-supply-problems. Maybe your cf-express-cardreader is better than mine but at the end, it could crash the same way.

stefans73, 2021-12-06:
Just tested: The ORICO 4-Port USB3.0 HUB works well!

thelightdecides, 2021-12-06:
That sounds good. I also did a few test runs, once it came to an error. The SSD and the card reader were both lost. Even after 60s they didn't show up again in the list, I unplugged them and plugged them back in, that didn't help either. Only a reboot of the Pi brought the two devices back on the list.
But all other test runs worked fine, even with the 2TB SSD from Sandisk, which previously did not work directly on the USB port of the Pi.
The new "remaining time" is very good. Also the whole web interface runs much more stable now in the new versions of LBB.

stefans73, 2021-12-06:
This means, you did all your tests with devices directly connected to the pi, without a hub? Did you recognize a re-connect of devices?
For now I'm very happy about this very small powered hub, it's moving into my backup-bag!

stefans73, 2021-12-08:
Is it correct that all memory cards are working now and disconnections only occur when the problematic USB devices are connected directly to the rpi? Does everything work properly with powerd usb hubs? If so, I would like to close this long issue.

thelightdecides, 2021-12-08:
Yes, I did all the test without a hub. Most of the time I use SD cards, and for SD cards and the Samsung T5 SDD, I don't need a hub. I only need the hub for my new Sandisk SSD and for the cf express cards. Often the cfexpress cards are copied without an error to the T5 SSD (both connected directly to the pi) but sometimes not. There was disconnections and reconnections of the cf express card reader (directly connectet to the pi). With both connected to the hub I got no error (during 7 test runs), only one when both devices get lost and no reconnection was possible.
I just copied 13000 files from the T5 SSD to the Sandisk SSD (both connected to the hub), no error. I think it is working now.

Thank you very much for all the work!!!

ruebyi, 2021-12-08:
So will you add your changes here, too?
The fix is not working for my SD Cards, yet - I will try the active USB hub and another Cardreader

[2 more replies]

closed 2022-01-13 by outdoorbits