outdoorbits/little-backup-box #121: SMB network share as storage location

andreweff, 2021-08-06:
Hei,

really nice work!

How to configure a SMB share as storage location?
I have a Synology NAS and want plugged in USB-Sticks get copied on the NAS.
I mounted the smb share with fstab in /mnt/networkshare but what should I set for STORAGE_DEV=? 

`STORAGE_DEV="**sda1**" # Name of the storage device
STORAGE_MOUNT_POINT="/mnt/networkshare" # Mount point of the storage device`

Thank you.

andreweff, 2021-08-06:
Hmm... Maybe I should try source to internal procedure...

stefans73, 2021-08-06:
Hi,

you just could point your file-browser to smb://IP.x.y.z and you will 
see the storages available. Actually this are:

smb://IP.x.y.z/internal
smb://IP.x.y.z/iOS
smb://IP.x.y.z/source
smb://IP.x.y.z/storage

Does this work for you?

Stefan

dmpop, 2021-08-06:
Please post general queries at https://discuss.pixls.us/c/hardware/little-backup-box/33 The **Issues** section  is for submitting bugs.

closed 2021-08-06 by dmpop