outdoorbits/little-backup-box #124: LBB Web UI not working

bernardoledo, 2021-08-13:
Hello there, I have just installed LBB in my Raspberry Pi 3, but for some reason I cannot access to the WebUI. 

While the raspberry is booting, I can see some error messages regarding services webui8000 and webui80, In the following lines you have more info on them.

Any thougth? 

Thanks! 
Bernardo Ledo.

`root@raspberrypi:/usr# systemctl status webui8000.service
● webui8000.service - web UI Port 8000
   Loaded: loaded (/etc/systemd/system/webui8000.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2021-08-13 14:56:19 BST; 9min ago
  Process: 3886 ExecStart=/usr/bin/php -S 0.0.0.0:8000 -t (code=exited, status=1/FAILURE)
 Main PID: 3886 (code=exited, status=1/FAILURE)

Aug 13 14:56:19 raspberrypi php[3886]:   --ri <name>      Show configuration for extension <name>.
Aug 13 14:56:19 raspberrypi systemd[1]: webui8000.service: Main process exited, code=exited, status=1/FAILURE
Aug 13 14:56:19 raspberrypi systemd[1]: webui8000.service: Failed with result 'exit-code'.
Aug 13 14:56:19 raspberrypi systemd[1]: webui8000.service: Service RestartSec=100ms expired, scheduling restart.
Aug 13 14:56:19 raspberrypi systemd[1]: webui8000.service: Scheduled restart job, restart counter is at 5.
Aug 13 14:56:19 raspberrypi systemd[1]: Stopped web UI Port 8000.
Aug 13 14:56:19 raspberrypi systemd[1]: webui8000.service: Start request repeated too quickly.
Aug 13 14:56:19 raspberrypi systemd[1]: webui8000.service: Failed with result 'exit-code'.
Aug 13 14:56:19 raspberrypi systemd[1]: Failed to start web UI Port 8000.
root@raspberrypi:/usr# systemctl status webui80.service
● webui80.service - web UI Port 80
   Loaded: loaded (/etc/systemd/system/webui80.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2021-08-13 14:47:52 BST; 17min ago
  Process: 526 ExecStart=/usr/bin/php -S 0.0.0.0:80 -t (code=exited, status=1/FAILURE)
 Main PID: 526 (code=exited, status=1/FAILURE)

Aug 13 14:47:51 raspberrypi php[526]:   --ri <name>      Show configuration for extension <name>.
Aug 13 14:47:51 raspberrypi systemd[1]: webui80.service: Main process exited, code=exited, status=1/FAILURE
Aug 13 14:47:51 raspberrypi systemd[1]: webui80.service: Failed with result 'exit-code'.
Aug 13 14:47:52 raspberrypi systemd[1]: webui80.service: Service RestartSec=100ms expired, scheduling restart.
Aug 13 14:47:52 raspberrypi systemd[1]: webui80.service: Scheduled restart job, restart counter is at 5.
Aug 13 14:47:52 raspberrypi systemd[1]: Stopped web UI Port 80.
Aug 13 14:47:52 raspberrypi systemd[1]: webui80.service: Start request repeated too quickly.
Aug 13 14:47:52 raspberrypi systemd[1]: webui80.service: Failed with result 'exit-code'.
Aug 13 14:47:52 raspberrypi systemd[1]: Failed to start web UI Port 80.
`

dmpop, 2021-08-13:
What happens when you run `systemctl start webui8000.service` or `systemctl start webui80.service`?  Same error?

stefans73, 2021-08-13:
Hi,

it works fine, you just have to do it as sudo. Your change revokes the 
following solution: comitup listens on port 80 while no network is 
connected. In comitup-config you can define a service to be 
stopped/restarted by comitup. webui80 is stopped by comitup while 
webui8000 is allway there and webui80 is restarted after comitup did its 
job. Anyway, your solution just renames the webui-service. If this 
works, the double solution works also (but integrates comitup well).

Stefan

---
Mit meinem angefügten öffentlichen Schlüssel "0xE61B756E.asc" können Sie 
gerne verschlüsselte Mails an mich senden. Näheres auch unter 
https://www.gpg4win.de/

Links:
------
[1] 
https://github.com/dmpop/little-backup-box/issues/124#issuecomment-898491846
[2] 
https://github.com/

bernardoledo, 2021-08-13:
> Hi, it works fine, you just have to do it as sudo. Your change revokes the following solution: comitup listens on port 80 while no network is connected. In comitup-config you can define a service to [...]

Hello there, thanks for you fast reply. 
I haven't installed comitup yet. I have just did a LBB greenfield installation.
Please let me know if you need any other information from my device to help me troubleshoot this issue.

BTW: I have basic Linux and Rapsberry Pi knowledge so I apologize in advance if I do some basic questions :)

bernardoledo, 2021-08-13:
> What happens when you run `systemctl start webui8000.service` or `systemctl start webui80.service`? Same error?

Same situation. Please check following lines: 

pi@raspberrypi:~ $ sudo su
root@raspberrypi:/home/pi#
root@raspberrypi:/home/pi#
root@raspberrypi:/home/pi# systemctl start webui8000.service
root@raspberrypi:/home/pi# systemctl status webui8000.service
● webui8000.service - web UI Port 8000
   Loaded: loaded (/etc/systemd/system/webui8000.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2021-08-13 17:57:32 BST; 4s ago
  Process: 32524 ExecStart=/usr/bin/php -S 0.0.0.0:8000 -t (code=exited, status=1/FAILURE)
 Main PID: 32524 (code=exited, status=1/FAILURE)

Aug 13 17:57:32 raspberrypi php[32524]:   --ri <name>      Show configuration for extension <name>.
Aug 13 17:57:32 raspberrypi systemd[1]: webui8000.service: Main process exited, code=exited, status=1/FAILURE
Aug 13 17:57:32 raspberrypi systemd[1]: webui8000.service: Failed with result 'exit-code'.
Aug 13 17:57:32 raspberrypi systemd[1]: webui8000.service: Service RestartSec=100ms expired, scheduling restart.
Aug 13 17:57:32 raspberrypi systemd[1]: webui8000.service: Scheduled restart job, restart counter is at 5.
Aug 13 17:57:32 raspberrypi systemd[1]: Stopped web UI Port 8000.
Aug 13 17:57:32 raspberrypi systemd[1]: webui8000.service: Start request repeated too quickly.
Aug 13 17:57:32 raspberrypi systemd[1]: webui8000.service: Failed with result 'exit-code'.
Aug 13 17:57:32 raspberrypi systemd[1]: Failed to start web UI Port 8000.
root@raspberrypi:/home/pi# date
Fri 13 Aug 17:57:38 BST 2021

stefans73, 2021-08-13:
Hi,

is that on a clean new installation or are there any cronjobs defined?

Are the crontabs empty for

sudo crontab -e

and

sudo -u pi crontab -e

?

Maybe there is already a process active listening on port 8000.

>

ruebyi, 2021-08-13:
I just tested this:
Corntab -e is epty

stefans73, 2021-08-13:
for both, root and pi?

bernardoledo, 2021-08-13:
> sudo -u pi crontab -e

Fresh new install. 

sudo crontab -e
 sudo -u pi crontab -e
@reboot sudo /home/pi/little-backup-box/scripts/backup.sh storage external >/dev/null 2>&1
@reboot sudo /home/pi/little-backup-box/scripts/restart-servers.sh
@reboot sudo /home/pi/little-backup-box/scripts/ip.sh

# Edit this file to introduce tasks to be run by cron.
#
# Each task to run has to be defined through a single line
# indicating with different fields when the task will be run
# and what command to run for the task
#
# To define the time you can provide concrete values for
# minute (m), hour (h), day of month (dom), month (mon),
# and day of week (dow) or use '*' in these fields (for 'any').
#
# Notice that tasks will be started based on the cron's system
# daemon's notion of time and timezones.
#
# Output of the crontab jobs (including errors) is sent through
# email to the user the crontab file belongs to (unless redirected).
#
# For example, you can run a backup of all your user accounts
# at 5 a.m every week with:
# 0 5 * * 1 tar -zcf /var/backups/home.tgz /home/
#
# For more information see the manual pages of crontab(5) and cron(8)
#
# m h  dom mon dow   command

ruebyi, 2021-08-13:
sudo -u pi crontab -e

@reboot sudo /home/pi/little-backup-box/scripts/backup.sh storage external >/de$
@reboot sudo /home/pi/little-backup-box/scripts/restart-servers.sh
@reboot sudo /home/pi/little-backup-box/scripts/ip.sh
@reboot sudo /home/pi/little-backup-box/scripts/start.sh

stefans73, 2021-08-13:
That doesn't explain it. But you should at least clean the root-crontab. 
crontab of root is from an older installation, right?. Best would be to 
have a new installation from scratch including actual raspbian.

ruebyi, 2021-08-13:
nope. fesh Install - I formatted everything

dmpop, 2021-08-13:
And you are using Raspberry Pi OS Lite, right?

ruebyi, 2021-08-13:
yes

ruebyi, 2021-08-13:
I only did an sudo update and Sudo upgrade bevor installing LBB

dmpop, 2021-08-13:
I have no idea what causes the problem. But I've came up with a possible fix. Right now it's in the `uuid` branch, but as soon as Stefan adds his changes and we run tests, I'll merge them into the `main` branch. Hopefully, this will fix the problem.

ruebyi, 2021-08-13:
My I ask how I could install that UUID Branch? Sorry, Feel like a github noob

dmpop, 2021-08-13:
There is no straightforward way to install from the `uuid` branch, I'm afraid. If you are willing to give it a go, here are the steps:

1. Connect to the Raspberry Pi via SSH.
2. Run the command `wget https://raw.githubusercontent.com/dmpop/little-backup-box/uuid/install-little-backup-box.sh`
3. Run the command `chmod +x install-little-backup-box.sh`
4. Open the installer file for editing with the command `nano install-little-backup-box.sh`
5. Add `git checkout uuid` line below line 81 `git clone https://github.com/dmpop/little-backup-box.git`
6. Save the changes
7. Run the installer using the command `./install-little-backup-box.sh`

It's late and I'm tired, so I hope I got all the steps.

ruebyi, 2021-08-13:
WebUI works in the UUID Branch

dmpop, 2021-08-13:
Fantastic! Thank you so much for testing and reporting back!

bernardoledo, 2021-08-13:
Thank you so much guys! It's up and runnning now

[image]

closed 2021-08-14 by dmpop