outdoorbits/little-backup-box #127: Auto mount now in RIPOs lite

glenerrington, 2022-01-06:
I have noticed today not just on Raspberry Pi OS that the OS is now mounting USB drives to/mount/"USB _UUID#" this renders the app useless unless I mount it to /media/storage in fstab by looks needs to be done.

I have noticed this with Linux Mint this morning has a HDD of mine that in the pass automounted to /media/glen/vault was mounted at /media/usb_UUID# which meant I had to rewrite my local shares after mounting it at the old mount point

outdoorbits, 2022-01-08:
I still think, you do not use the correct platform. Please prepare your 
sd-card with Raspberry Pi Imager (https://www.raspberrypi.com/software/) 
and select at first Raspberry Pi OS (other) and then Raspberry Pi OS 
Lite (32-bit). Please use exactly that, this is the platform lbb is 
developed on an running.

glenerrington, 2022-01-08:
I am not a noob at the whole computer this yes I am running raspberry pi OS lite. and I know how to repair an install nation of PiOS

for me I have a big issue with the way this has be developed as there is a backup photo software for mobile phones that run iOs and Android and with the mothiod that you are using to mount the external storage is not helpful to users who may want to use their Raspberry Pi for more than LBB the software for mobile backing up is https://github.com/lomorage/homepage

Also I have just got a SD card reader branded Integral (https://smile.amazon.co.uk/gp/product/B0047T6XWY/) that for some reason does not show up properly in the lite version even blkid command can not get UUIDs for.

In my opinion your software needs to be more flexible with how it can be used especially for the photography market

outdoorbits, 2022-01-09:
If you want to have a backup-solution running parallel to other USB-applications at the same time, of course lbb will not be what you are looking for. On the other hand side, to use your raspberry in multiple modes, you can provide a sd for each task.
If you need help in the standard-configuration, you are welcome.

closed 2022-01-09 by outdoorbits