outdoorbits/little-backup-box #128: SD card to USB drive backup fails

jeroenzelle, 2022-01-07:
**Describe the bug**
I am testing the **source** (SD card in usb reader) to **external storage** (USB drive) function, however getting errors. It immediately says: Lost device . 

However when I try it the other way around, so with files on the usb drive and storing it to the sd card ... then it does work initally without errors. But after a while , I do get the same error.

** DEVICES USED **
- Odroid UX4 (your script does work fine on it) with DietPi
- Lexar USB 3.0 card reader (2 different types)
- Angelbird 64gb SD card

**See logfile below for the error when copying from sd card to usb drive:**

> `
> 22:34:55
> No abnormal conditions have been detected
> 
> 22:34:54
> Lost device '--uuid 29C4-ED50': DEVICE LOST
> 
> 22:34:54
> /var/www/little-backup-box/tmp/sync.log: >>>>>
> 2022/01/07 22:34:54 [8219] building file list
> 2022/01/07 22:34:54 [8219] .d..t.og... ./
> 2022/01/07 22:34:54 [8219] Number of files: 1 (dir: 1)
> 2022/01/07 22:34:54 [8219] Number of created files: 0
> 2022/01/07 22:34:54 [8219] Number of deleted files: 0
> 2022/01/07 22:34:54 [8219] Number of regular files transferred: 0
> 2022/01/07 22:34:54 [8219] Total file size: 0 bytes
> 2022/01/07 22:34:54 [8219] Total transferred file size: 0 bytes
> 2022/01/07 22:34:54 [8219] Literal data: 0 bytes
> 2022/01/07 22:34:54 [8219] Matched data: 0 bytes
> 2022/01/07 22:34:54 [8219] File list size: 0
> 2022/01/07 22:34:54 [8219] File list generation time: 0.003 seconds
> 2022/01/07 22:34:54 [8219] File list transfer time: 0.000 seconds
> 2022/01/07 22:34:54 [8219] Total bytes sent: 66
> 2022/01/07 22:34:54 [8219] Total bytes received: 19
> 2022/01/07 22:34:54 [8219] sent 66 bytes  received 19 bytes  170.00 bytes/sec
> 2022/01/07 22:34:54 [8219] total size is 0  speedup is 0.00
> <<<<<
> 
> 22:34:53
> Source o.k.
> Working...
> size: 14G
> used: 7%
> 
> 22:34:52
> mount_device RESULT=--uuid 29C4-ED50
> 
> 22:34:50
> DEVICE_IDENT_THIS='--uuid 29C4-ED50' ('/dev/sda1') prepared to mount at '/media/source'
> 
> 22:34:34
> Ready
> Insert source
> Ext. storage o.k.
> size: 15G
> 
> 22:34:30
> Ext. storage o.k.
> size: 15G
> free: 15G
> Ready
> 
> 22:34:30
> mount_device RESULT=--uuid 56BE650ABE64E449
> 
> 22:34:28
> DEVICE_IDENT_THIS='--uuid 56BE650ABE64E449' ('/dev/sdb1') prepared to mount at '/media/storage'
> `

And the logfile when copying from usb drive to sd card reader:

> `23:39:58
> Backup status:
> 1 of 51
> 1%           00:02:30
> 
> 
> 23:39:55
> Backup status:
> 0 of 51
> 0%                   
> 
> 
> 23:39:54
> mount_device RESULT=--uuid 40F9-6C3D
> 
> 23:39:50
> DEVICE_IDENT_THIS='--uuid 40F9-6C3D' ('/dev/sdc1') prepared to mount at '/media/storage'
> 
> 23:39:44
> Try backup 2 of 5
> Backup status:
> 37 of 51
> 72%          00:01:02
> 
> 23:39:42
> No abnormal conditions have been detected
> 
> 23:39:42
> Lost device '--uuid 40F9-6C3D': DEVICE LOST
> 
> 23:39:42
> /var/www/little-backup-box/tmp/sync.log: >>>>>
> 2022/01/07 23:36:54 [16854] building file list
> 2022/01/07 23:36:54 [16854] .d..t...... ./
> 2022/01/07 23:36:54 [16854] cd+++++++++ DCIM/
> 2022/01/07 23:36:54 [16854] cd+++++++++ DCIM/111LEICA/
> 2022/01/07 23:37:01 [16854] >f+++++++++ DCIM/111LEICA/L1110698.DNG
> 2022/01/07 23:37:06 [16854] >f+++++++++ DCIM/111LEICA/L1110699.DNG
> 2022/01/07 23:37:11 [16854] >f+++++++++ DCIM/111LEICA/L1110700.DNG
> 2022/01/07 23:37:15 [16854] >f+++++++++ DCIM/111LEICA/L1110701.DNG
> 2022/01/07 23:37:19 [16854] >f+++++++++ DCIM/111LEICA/L1110702.DNG
> 2022/01/07 23:37:23 [16854] >f+++++++++ DCIM/111LEICA/L1110703.DNG
> 2022/01/07 23:37:28 [16854] >f+++++++++ DCIM/111LEICA/L1110704.DNG
> 2022/01/07 23:37:32 [16854] >f+++++++++ DCIM/111LEICA/L1110705.DNG
> 2022/01/07 23:37:37 [16854] >f+++++++++ DCIM/111LEICA/L1110706.DNG
> 2022/01/07 23:37:41 [16854] >f+++++++++ DCIM/111LEICA/L1110707.DNG
> 2022/01/07 23:37:46 [16854] >f+++++++++ DCIM/111LEICA/L1110708.DNG
> 2022/01/07 23:37:50 [16854] >f+++++++++ DCIM/111LEICA/L1110709.DNG
> 2022/01/07 23:37:54 [16854] >f+++++++++ DCIM/111LEICA/L1110710.DNG
> 2022/01/07 23:37:59 [16854] >f+++++++++ DCIM/111LEICA/L1110711.DNG
> 2022/01/07 23:38:03 [16854] >f+++++++++ DCIM/111LEICA/L1110712.DNG
> 2022/01/07 23:38:07 [16854] >f+++++++++ DCIM/111LEICA/L1110713.DNG
> 2022/01/07 23:38:12 [16854] >f+++++++++ DCIM/111LEICA/L1110714.DNG
> 2022/01/07 23:38:16 [16854] >f+++++++++ DCIM/111LEICA/L1110715.DNG
> 2022/01/07 23:38:21 [16854] >f+++++++++ DCIM/111LEICA/L1110716.DNG
> 2022/01/07 23:38:25 [16854] >f+++++++++ DCIM/111LEICA/L1110717.DNG
> 2022/01/07 23:38:29 [16854] >f+++++++++ DCIM/111LEICA/L1110718.DNG
> 2022/01/07 23:38:34 [16854] >f+++++++++ DCIM/111LEICA/L1110719.DNG
> 2022/01/07 23:38:38 [16854] >f+++++++++ DCIM/111LEICA/L1110720.DNG
> 2022/01/07 23:38:44 [16854] >f+++++++++ DCIM/111LEICA/L1110721.DNG
> 2022/01/07 23:38:48 [16854] >f+++++++++ DCIM/111LEICA/L1110722.DNG
> 2022/01/07 23:38:52 [16854] >f+++++++++ DCIM/111LEICA/L1110723.DNG
> 2022/01/07 23:38:57 [16854] >f+++++++++ DCIM/111LEICA/L1110724.DNG
> 2022/01/07 23:39:01 [16854] >f+++++++++ DCIM/111LEICA/L1110725.DNG
> 2022/01/07 23:39:05 [16854] >f+++++++++ DCIM/111LEICA/L1110726.DNG
> 2022/01/07 23:39:10 [16854] >f+++++++++ DCIM/111LEICA/L1110727.DNG
> 2022/01/07 23:39:14 [16854] >f+++++++++ DCIM/111LEICA/L1110728.DNG
> 2022/01/07 23:39:18 [16854] >f+++++++++ DCIM/111LEICA/L1110729.DNG
> 2022/01/07 23:39:22 [16854] >f+++++++++ DCIM/111LEICA/L1110730.DNG
> 2022/01/07 23:39:27 [16854] >f+++++++++ DCIM/111LEICA/L1110731.DNG
> 2022/01/07 23:39:31 [16854] >f+++++++++ DCIM/111LEICA/L1110732.DNG
> 2022/01/07 23:39:36 [16854] >f+++++++++ DCIM/111LEICA/L1110733.DNG
> 2022/01/07 23:39:42 [16854] >f+++++++++ DCIM/111LEICA/L1110734.DNG
> 2022/01/07 23:39:42 [16854] cd+++++++++ PRIVATE/
> 2022/01/07 23:39:42 [16854] >f+++++++++ PRIVATE/FASTLOAD.DAT
> 2022/01/07 23:39:42 [16854] >f+++++++++ PRIVATE/META_001.DAT
> 2022/01/07 23:39:42 [16854] >f+++++++++ PRIVATE/META_001.TMP
> 2022/01/07 23:39:42 [16854] >f+++++++++ PRIVATE/META_002.DAT
> 2022/01/07 23:39:42 [16854] >f+++++++++ PRIVATE/META_002.TMP
> 2022/01/07 23:39:42 [16854] cd+++++++++ PRIVATE/TEMP/
> 2022/01/07 23:39:42 [16854] >f+++++++++ PRIVATE/TEMP/0b0b99a0.CPC
> 2022/01/07 23:39:42 [16854] >f+++++++++ PRIVATE/TEMP/0b0b99a0.CPG
> 2022/01/07 23:39:42 [16854] cd+++++++++ System Volume Information/
> 2022/01/07 23:39:42 [16854] >f+++++++++ System Volume Information/WPSettings.dat
> 2022/01/07 23:39:42 [16854] Number of files: 51 (reg: 45, dir: 6)
> 2022/01/07 23:39:42 [16854] Number of created files: 50 (reg: 45, dir: 5)
> 2022/01/07 23:39:42 [16854] Number of deleted files: 0
> 2022/01/07 23:39:42 [16854] Number of regular files transferred: 45
> 2022/01/07 23:39:42 [16854] Total file size: 3.22G bytes
> 2022/01/07 23:39:42 [16854] Total transferred file size: 3.22G bytes
> 2022/01/07 23:39:42 [16854] Literal data: 3.22G bytes
> 2022/01/07 23:39:42 [16854] Matched data: 0 bytes
> 2022/01/07 23:39:42 [16854] File list size: 0
> 2022/01/07 23:39:42 [16854] File list generation time: 0.003 seconds
> 2022/01/07 23:39:42 [16854] File list transfer time: 0.000 seconds
> 2022/01/07 23:39:42 [16854] Total bytes sent: 3.23G
> 2022/01/07 23:39:42 [16854] Total bytes received: 906
> 2022/01/07 23:39:42 [16854] sent 3.23G bytes  received 906 bytes  19.14M bytes/sec
> 2022/01/07 23:39:42 [16854] total size is 3.22G  speedup is 1.00
> <<<<<
> 
> 23:39:41
> Backup status:
> 37 of 51
> 72%          00:01:02
> >>>>>>>>>>>>>>>`

outdoorbits, 2022-01-08:
This seems to me to be a known problem. But I guess, it's a hardware-problem about the power supply. Maybe it's working if you just use another (better) power adapter. In my case, I got stable conditions on my powerbank (Anker ~2700 mAh) or using a powered hub. Please have a look at https://github.com/outdoorbits/little-backup-box/wiki/02.-Hardware at the bottom of the file you will find Useful accessories and below that Travel hubs.

ruebyi, 2022-01-29:
I'm not sure wether this is helpful:

I ran into the same error today - with a card reader and SSD connected to my USB Hub.
The Card failed several Times until I deleted the ID file.
After that the card was read out with a new ID and the Backup completed - with a card reader and SSD connected to my USB Hub.

outdoorbits, 2022-01-29:
Your first message was much more easy. ;-) Thank you for both!
I will add your experience soon.

About the problem with the id-file: If you run into this trouble again, 
could you please set loglevel to 3 and send your logging?Message ID: 
***@***.***>

closed 2022-02-28 by outdoorbits