outdoorbits/little-backup-box #129: Error with copy from SD mobile

Asturbike, 2022-01-11:
Hi,

I have a Xioami Redmi Note 9 Pro and when it comes to backup I have not been able to configure it.
If I select copy from camera to internal I make the copy of the DCIM folder of the mobile but my photos are on the SD card.

I have tried to make a copy of the SD by selecting source - internal storage but it does not find the SD of the mobile.

Any configuration to copy the folders from the SD card of the mobile?

I have tried to select the path in the directory settings for backup but nothing.

Thank you very much for your help.

Translated with www.DeepL.com/Translator (free version)

outdoorbits, 2022-01-11:
Hi,
This is probably due to the (still) missing documentation, sorry. Most smartphones must be switched to PTP mode first. Has that already happened?
In the event that this is not the problem, I have just expanded the logging. Please perform Settings -> update.
Did that help?

Asturbike, 2022-01-11:
Hi,
Finally works fine too!!
The problem, maybe, is that the SD have a lot of documents and directories (or the SD size.. or OrangePi Zero speed..) and the backup takes 5 minutes to start.
If I wait 5 minutes I see the backup in course.
Thanks.

I try to update and the screen is all white..
[image]

outdoorbits, 2022-01-11:
From my Galaxy A72 it takes also a while to read the folders, but maybe 15, not 300 seconds! But of course, it might be your large sd-card. Is the backup-speed acceptable?

I just did a update (to test) without problems. If it does not start, you could try a reload... But for to update you also can run the install-script like you did in the installation-process.

Asturbike, 2022-01-12:
I have 2 phones and the "problem" is the same, about 300 secs to start, but the backup-speed is normal.
I update with the script with the installation.

Thanks , your work is fantastic!!

If you need help to translations to spanish or any task about lbb, contact me!

outdoorbits, 2022-01-12:
Thank you. I think, if you pull up your log level to 3 you would see hundreds of folders and that's why it takes such a long time.

outdoorbits, 2022-01-12:
Hi, I just added a function to speed up a lot and save that  minutes. Please update and have a look at https://github.com/outdoorbits/little-backup-box/wiki/04.-Setup

Asturbike, 2022-01-13:
Hi,
Now not work, with the new rute RedmiNote9Pro:!/store_00020002/DCIM or *:DCIM;RedmiNote9Pro:!/store_00020002/DCIM or *:DCIM
The mail says "backup succesful" but the directory is empty, only this directories....
[image]

Log:
opia de seguridad completa.

Escribe: camera a internal 
Camera: RedmiNote9Pro

Registro de respaldo:

---- tratar 1 ----

1 intentar/intentos necesarios.

07:43:33
/var/www/little-backup-box/tmp/sync.log:
>>>>>
0.000066 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000334 main                        (2): gphoto2 2.5.27
0.000375 main                        (2): gphoto2 has been compiled with the following options:
0.000401 main                        (2):  + gcc (C compiler used)
0.000424 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000445 main                        (2):  + exif (for displaying EXIF information)
0.000467 main                        (2):  + cdk (for accessing configuration options)
0.000488 main                        (2):  + aa (for displaying live previews)
0.000511 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000531 main                        (2):  + readline (for easy navigation in the shell)
0.000582 main                        (2): libgphoto2 2.5.27
0.000615 main                        (2): libgphoto2 has been compiled with the following options:
0.000642 main                        (2):  + standard camlib set (ax203 canon digigr8 dimagev directory docupen jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801)
0.000667 main                        (2):  + gcc (C compiler used)
0.000687 main                        (2):  + ltdl (for portable loading of camlibs)
0.000707 main                        (2):  + EXIF (for special handling of EXIF files)
0.000730 main                        (2): libgphoto2_port 0.12.0
0.000753 main                        (2): libgphoto2_port has been compiled with the following options:
0.000775 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000795 main                        (2):  + gcc (C compiler used)
0.000815 main                        (2):  + ltdl (for portable loading of iolibs)
0.000835 main                        (2):  + EXIF (for vusb)
0.000856 main                        (2):  + USB (libusb1, for USB cameras)
0.000876 main                        (2):  + serial (for serial cameras)
0.000896 main                        (2):  + no resmgr (serial port access and locking)
0.000916 main                        (2):  + no ttylock (serial port locking)
0.000936 main                        (2):  + no lockdev (serial port locking)
0.000958 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000979 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000999 main                        (2): invoked with following arguments:
0.001021 main                        (2):   --filename
0.001043 main                        (2):   %F/%f.%C
0.001063 main                        (2):   --get-all-files
0.001083 main                        (2):   --folder
0.001103 main                        (2):   /store_00020002/DCIM
0.001124 main                        (2):   --skip-existing
0.001145 main                        (2):   --list-files
0.001165 main                        (2):   --debug
0.001185 main                        (2):   --debug-logfile
0.001206 main                        (2):   /var/www/little-backup-box/tmp/sync.log
0.001535 load_settings               (2): Creating gphoto config directory ('/root/.gphoto')
0.001740 load_settings               (2): Loading settings from file '/root/.gphoto/settings'.
0.002895 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.002976 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/lib/arm-linux-gnueabihf/libgphoto2_port/0.12.0'...
0.003410 foreach_func                (2): Called for filename '/usr/lib/arm-linux-gnueabihf/libgphoto2_port/0.12.0/disk'.
0.004336 gp_port_library_list        (2): found fstab fsname UUID=8c826ac0-fd4b-4447-b3ce-0c304b6566ea
0.004429 gp_port_library_list        (2): found fstab fsname tmpfs
0.004894 gp_port_library_list        (2): found mtab fsname /dev/mmcblk0p1
0.005010 gp_port_library_list        (2): found mtab fsname udev
0.005208 gp_port_library_list        (2): found mtab fsname devpts
[... 620 lines cut ...]
0.530806 delete_all_files            (2): Delete all files in folder 0x830b30/store_00020001
0.530836 recurse_delete_folder       (2): Recurse delete folder 0x81df00/store_00010001
0.530863 delete_folder               (2): Delete one folder 0x81df00/store_00010001
0.530892 delete_all_files            (2): Delete all files in folder 0x81df00/store_00010001
0.530923 delete_all_files            (2): Delete all files in folder 0x8126c0//
0.530951 gp_port_free                (2): Freeing port...
0.530980 gp_port_close               (2): Closing port...
0.532315 gp_filesystem_reset         (2): resetting filesystem
0.532447 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.532479 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.532504 delete_all_folders          (2): Internally deleting all folders from '/'...
[... cut ...] up folder '/'...
0.532561 lookup_folder               (2): Found! / is 0x8126c0
0.532585 recurse_delete_folder       (2): Recurse delete folder 0x8126c0//
0.532614 delete_all_files            (2): Delete all files in folder 0x8126c0//
<<<<<

07:43:31
 - /store_00020002/DCIM - For use as pattern in Settings ('Directorios para respaldar en teléfonos inteligentes y cámaras'): 'RedmiNote9Pro:!/store_00020002/DCIM'

07:43:31
Folders to sync from camera 'RedmiNote9Pro':

07:43:30
Camera: RedmiNote9Pro

outdoorbits, 2022-01-13:
For a quick try, could you disable Log-sync-protocol in Settings?

Asturbike, 2022-01-13:
the problem continue, a lot of tmpfiles and directories and the copy is slow, same cable and same phone.
Tis backup yesterday 10 minutes, today...

[image]

outdoorbits, 2022-01-13:
There is an update online now. For me this was working now. Does it solve the problem in your environment too?

Asturbike, 2022-01-13:
Yes!!
Time es normal, very fast.
[image]

But now is the same case that yesterday, I wait 200 o 300 secs and the backup starts, but I am in no hurry!!

Thanks!!!

One question... Is it possible to have a extra button to copy internal storage (/media/internal/*) to external storage?
I use for this project one orange pi zero with only one USB, and my idea is to cpy the photos to internal SD and later copy this photos to Nextcloud, this option works fine, but could copy other backup to the pendrive....

outdoorbits, 2022-01-13:
Of course, good idea. The additional button was not a big deal and is available now.
But in my tests the folders are not scanned anymore and backup from smartphone starts immediately. Could you please send another logfile of a backup having such a delay?

Asturbike, 2022-01-13:
Now I tested the new button, Int to Ext, works very good!! Its very important for me thanks!!

Tomorrow I Will delete the phone directory and i Will test the backup and Will send the log

Asturbike, 2022-01-14:
[image]

in the options I only have set to check the sd, but it also tries to copy from internal memory.
When I activate the copy, it sends a copy completed email, then the usual wait and then it tries to copy from the SD.

log:

08:38:01
IP (online):
192.168.0.234
Backup complete.
Do not unplug!

08:37:42
Backup complete.
Do not unplug!
Power down via
web UI

08:37:41
Mail:
Little Backup Box: Backup Backup complete.

Backup complete.

Type: camera to internal 
Camera: RedmiNote9Pro

Backup log:

---- try 1 ----
Skip existing file store_00010001/DCIM/.tmfs/.turing.dat
Skip existing file store_00010001/DCIM/.tmfs/sk_v.dat
Skip existing file store_00010001/DCIM/.tmsdual/860300598/sk_g.dat
Skip existing file store_00010001/DCIM/Creative/IMG_20210627_211724.jpg
Skip existing file store_00010001/DCIM/Creative/IMG_20210627_211816.jpg
.................
.................
.................
xxxxxxxxxxxxxxxxxxxxxxxxx
delete for the log.....
xxxxxxxxxxxxxxxxxxxxxxxx

1 try/tries needed.

08:37:36
Backup status:
1 of 699
0%           00:34:54

08:37:33
Backup status:
0 of 699
%                    
comparing...

08:37:29
 - /store_00020001/DCIM - For use as pattern in Settings ('Directories to be backed up on smartphones and cameras'): 'RedmiNote9Pro:!/store_00020001/DCIM'

08:37:29
 - /store_00010001/DCIM - For use as pattern in Settings ('Directories to be backed up on smartphones and cameras'): 'RedmiNote9Pro:!/store_00010001/DCIM'

08:37:29
Folders to sync from camera 'RedmiNote9Pro':

08:34:01
IP (online):
192.168.0.234
Backup complete.
Do not unplug!

08:33:42
Backup complete.
Do not unplug!
Power down via
web UI

08:33:41
Mail:
Little Backup Box: Backup Backup complete.

Backup complete.

Type: camera to internal 
Camera: 

Backup log:

---- try 1 ----
For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --filename %F/%f.%C --get-all-files --folder / --skip-existing --list-files

Please make sure there is sufficient quoting around the arguments.

1 try/tries needed.

08:33:35
 - / - For use as pattern in Settings ('Directories to be backed up on smartphones and cameras'): ':!/'

08:33:35
Folders to sync from camera '':

08:33:34
Camera: 

08:33:34
Camera o.k.
Working...
Ready

08:33:30
Camera o.k.
Working...
Ready
Connect camera

08:33:26
Ready
Connect camera
Int. storage o.k.
size: 113G

08:33:23
Int. storage o.k.
size: 113G
free: 97G
RedmiNote9Pro

08:33:22
Destination: internal 

08:33:22
Source: camera

08:33:12
Camera: RedmiNote9Pro

08:33:12
RedmiNote9Pro
Camera o.k.
Working...
IP (online):

08:33:08
Camera o.k.
Working...
IP (online):
192.168.0.234

outdoorbits, 2022-01-14:
Hi, could you please update and send the entry of your settings field?

Asturbike, 2022-01-14:
What configuration do you need?

[image]

outdoorbits, 2022-01-14:
Yes, this one, thank you. Same problem after update? If yes, could you send your log again?

Asturbike, 2022-01-14:
I have 4 new photos but the lbb said backup complete...

09:52:54
Source: camera

09:52:54
Destination: internal 

09:52:55
Int. storage o.k.
size: 113G
free: 97G
IP (online):

09:52:59
Ready
Connect camera
Int. storage o.k.
size: 113G

09:53:03
IP (online):
192.168.0.234
Ready
Connect camera

09:53:03
Camera o.k.
Working...
IP (online):
192.168.0.234

09:53:07
RedmiNote9Pro
Camera o.k.
Working...
IP (online):

09:53:07
Camera: RedmiNote9Pro

09:53:07
Folders to sync from camera 'RedmiNote9Pro':

09:53:07
 - /store_00020002/DCIM - For use as pattern in Settings ('Directories to be backed up on smartphones and cameras'): 'RedmiNote9Pro:!/store_00020002/DCIM'

09:53:12
Mail:
Little Backup Box: Backup Backup complete.

Backup complete.

Type: camera to internal 
Camera: RedmiNote9Pro

Backup log:

---- try 1 ----
For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --filename %F/%f.%C --get-all-files --folder /store_00020002/DCIM --skip-existing --list-files

Please make sure there is sufficient quoting around the arguments.

1 try/tries needed.

09:53:13
Backup complete.
Do not unplug!
Power down via
web UI

09:54:01
IP (online):
192.168.0.234
Backup complete.
Do not unplug!

outdoorbits, 2022-01-14:
gphoto2 does not get a files count but gets the phones name. I had this when my phone didn't accept the connection. Was your phone unlocked? Did it ask to permit access?

Asturbike, 2022-01-14:
yes my phone is unlocked and... i don´t know if permit access, if the setting is *DCIM the copy is ok

outdoorbits, 2022-01-14:
Could you send the Log of *:DCIM please? I will add a few log-points later today and ask you again for logs...

outdoorbits, 2022-01-14:
Of course, the file-backup runs as root (sudo), the owner is o.k.
Did you check the destination-folder for the files? Maybe they are already there? On camera-backup the script just gets the number of images at the camera but it can't compare with the target. So it tries to copy all the images to target but at target the filecount doesn't change, because the images have already been there. So the progress-information fails...

outdoorbits, 2022-01-14:
... and there is an update with some more log-points around camera-backup. Could you try a backup RedmiNote9Pro:!/store_00020002/DCIM and send the logfile (loglevel 3)?

Asturbike, 2022-01-17:
don´t found new photos...

> ... and there is an update with some more log-points around camera-backup. Could you try a backup RedmiNote9Pro:!/store_00020002/DCIM and send the logfile (loglevel 3)?

13:53:52
Source: camera

13:53:52
Destination: internal 

13:53:52
umount_device: 'usb_1'? ''

13:53:52
umount_device: 'usb_2'? ''

13:53:53
Int. storage o.k.
size: 113G
free: 97G
Backup complete.

13:53:57
Ready
Connect camera
Int. storage o.k.
size: 113G

13:54:00
Camera o.k.
Working...
Ready
Connect camera

13:54:03
IP (online):
192.168.0.234
Camera o.k.
Working...

13:54:04
RedmiNote9Pro
IP (online):
192.168.0.234
Camera o.k.

13:54:04
Camera: RedmiNote9Pro

13:54:04
Folders to sync from camera 'RedmiNote9Pro':

13:54:05
 - /store_00020002/DCIM - For use as pattern in Settings ('Directories to be backed up on smartphones and cameras'): 'RedmiNote9Pro:!/store_00020002/DCIM'

13:54:06
gphoto2 --list-files --folder "/store_00020002/DCIM":
exitcode=1
For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --list-files --folder /store_00020002/DCIM

Please make sure there is sufficient quoting around the arguments.

13:54:06
Files in folder '/store_00020002/DCIM': 0

13:54:06
Files to sync before backup: 0

13:54:06
Backup from camera: /store_00020002/DCIM

13:54:06
SYNC_RETURN_CODE: 1; SYNC_TIME: 0

13:54:07
umount_device: 'usb_1'? ''

13:54:07
umount_device: 'usb_2'? ''

13:54:10
Mail:
Little Backup Box: Backup Backup complete.

Backup complete.

Type: camera to internal 
Camera: RedmiNote9Pro

Backup log:

---- try 1 ----
For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --filename %F/%f.%C --get-all-files --folder /store_00020002/DCIM --skip-existing --list-files

Please make sure there is sufficient quoting around the arguments.

1 try/tries needed.

13:54:11
Backup complete.
Do not unplug!
Power down via
web UI

13:55:07
IP (online):
192.168.0.234
Backup complete.
Do not unplug!

outdoorbits, 2022-01-17:
We are getting closer. At least we have an error message: 
gphoto2 --list-files --folder "/store_00020002/DCIM"
causes an error. Could you please run a "*:DCIM" backup on the same device, with the same sd-card again and post the log? Alternatively you can also login via ssh and execute this:
`gphoto2 --list-files --folder | grep DCIM`
I hope, the paths differ to "/store_00020002/DCIM".

There is one difference between your last post and your last *:DCIM logfile
In your last post. You configured
"RedmiNote9Pro:!/store_00020002/DCIM"
while before you had
"RedmiNote9Pro:!/store_00020001/DCIM"

Maybe it is just that? What does the 0002000**1**/DCIM and 0002000**2**/DCIM mean?
You could configure both (if you use 2 sd-cards in exchange - maybe the difference indicates the cards?):
RedmiNote9Pro:!/store_00020001/DCIM;RedmiNote9Pro:!/store_00020002/DCIM
should check both options.

Asturbike, 2022-01-18:
Yes!! My error, Really is  /media/internal/RedmiNote9Pro/store_00020001/DCIM/Camera/

Now works but process no finish

13:29:52
Source: camera

13:29:52
Destination: internal 

13:29:52
umount_device: 'usb_1'? ''

13:29:52
umount_device: 'usb_2'? ''

13:29:53
Int. storage o.k.
size: 113G
free: 97G
IP (online):

13:29:57
Ready
Connect camera
Int. storage o.k.
size: 113G

13:30:01
IP (online):
192.168.0.234
Ready
Connect camera

13:30:22
Camera o.k.
Working...
IP (online):
192.168.0.234

13:30:26
RedmiNote9Pro
Camera o.k.
Working...
IP (online):

13:30:26
Camera o.k.
Working...
Camera o.k.
Working...

13:30:26
Camera: RedmiNote9Pro

13:30:26
Folders to sync from camera 'RedmiNote9Pro':

13:30:26
 - /store_00020002/DCIM - For use as pattern in Settings ('Directories to be backed up on smartphones and cameras'): 'RedmiNote9Pro:!/store_00020002/DCIM'

13:30:27
gphoto2 --list-files --folder "/store_00020002/DCIM":
exitcode=1
For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --list-files --folder /store_00020002/DCIM

Please make sure there is sufficient quoting around the arguments.

13:30:27
Files in folder '/store_00020002/DCIM': 0

13:30:27
Files to sync before backup: 0

13:30:27
Backup from camera: /store_00020002/DCIM

13:30:28
SYNC_RETURN_CODE: 1; SYNC_TIME: 1

13:30:28
umount_device: 'usb_1'? ''

13:30:28
umount_device: 'usb_2'? ''

13:30:29
RedmiNote9Pro
Backup status:
0 of 0
?%                   

13:30:29
Camera: RedmiNote9Pro

13:30:29
Folders to sync from camera 'RedmiNote9Pro':

13:30:29
 - /store_00020001/DCIM - For use as pattern in Settings ('Directories to be backed up on smartphones and cameras'): 'RedmiNote9Pro:!/store_00020001/DCIM'

13:30:35
gphoto2 --list-files --folder "/store_00020001/DCIM":
exitcode=0
There is no file in folder '/store_00020001/DCIM'.
There are 706 files in folder '/store_00020001/DCIM/Camera'.
#1     IMG_20210927_201554.jpg    rd  7363 KB 4640x3472 image/jpeg
#2     IMG_20210929_105736.jpg    rd  4906 KB 4640x3472 image/jpeg
************************************
********************************

13:30:35
Files in folder '/store_00020001/DCIM': 706

13:30:35
Files to sync before backup: 706

13:30:36
Backup from camera: /store_00020001/DCIM

13:30:36
Backup status:
0 of 706
%                    
comparing...

13:30:39
Backup status:
4 of 706
0%           00:08:46

13:30:54
SYNC_RETURN_CODE: 0; SYNC_TIME: 19

13:30:55
umount_device: 'usb_1'? ''

13:30:55
umount_device: 'usb_2'? ''

outdoorbits, 2022-01-18:
Hi, well, last post wasn't loglevel 3. right? But it looks good. Are you sure, the target folder was empty before? Could it be, the files have been already there? The script detects 706 files on source. For me it looks like, it checks the files and only 4 of them has been new. Did the logfile really end with the line _umount_device: 'usb_2'? ''"_?

outdoorbits, 2022-01-21:
I'm very interested about every potential problem of lbb, so descriptions of trouble are important to me. It's the only way, to make it "bullet proof" one day. ;-)
Can you still reproduce the problem? Thank you for your help on debugging!

closed 2022-02-28 by outdoorbits