outdoorbits/little-backup-box #132: No able to get the script to run

dander45, 2022-01-13:
I am new to Raspberry Pi so I apologise for any basic questions.
I am trying to load and run the LBB as it appears to be something I make use of.
Following the instructions I have downloaded Raspberry PI OS Lite (32-BIT) to a new SD card and activated SSH in the Imager
I have loaded this into my Raspberry Pi 4 and it goes through a load of uploading etc finally waiting for me to login to Raspberry Pi
I do this and re-boot, it then suggests I change my password, which I do.
then I run the cmd as in the Readme file
curl -sSL https://raw.githubusercontent.com/outdoorbits/little-backup-box/main/install-little-backup-box.sh | bash 2> install-error.log
the script starts to run....
asks for default mode, I select NONE
ask to install mejiro I say yes
ask to install comitup I say yes
It then does a lot of lines of script
It goes through a progress bar at the bottom, no apparent errors

After about 20 mins of code going up the screen it gets to another progress bar across the bottom,
mostly selecting - preparing - unpacking
then creates a number of config files
and a pile of setting up
it says home directory cannot be created - couldn't read it all, it went up too quickly
it then does some downloading
does some more get
and another progress bar at the bottom
then installs mejiro
then another progress bar - quite quick thi sone
more gets
another progress bar
then screen goes blank
and a good number of lines then pass starting with OK
then waits for login
inpup login and system sits at command line prompt.
no errors visible
not sure if I am supposed to run something now or if the installation has failed - any suggestions
in total this took 50minutes, is that about right?

outdoorbits, 2022-01-13:
Looks good, but extremely slow. Please just try to access the box by browser: https://IP.OF.YOUR.BOX

outdoorbits, 2022-01-15:
... or http://IP.OF.YOUR.BOX:8000

dander45, 2022-01-17:
Hi, Yes I can get on to the app from my pc using https://myip  but I think I am having problems with the USB drive I am using for the backup so I I decided it was better to replace this before I go much further, should be in the next day or two.

Thanks

David.

David Anderson
Acoustic Engineer
Hebden Sound Ltd.
172 Wadsley lane,
Sheffield
S6 4EE
07774 782503
[facebook sml]
Visit us on Facebook<http://www.facebook.com/accusound.microphone.systems>
Bridge Microphones and Accusound are trading names of Hebden Sound Ltd.

dander45, 2022-01-20:
Hi,
I now have a new SSD drive and SD card.
The faster card means LLB now downloads and installs in less than 10 minutes, which is much better and I feel was some of the problem.
The USB SSD storage is also faster and works reliably on PC.

I downloaded and ran LBB first time and it worked fine, I have added the log file. Little-backup-box.log first use.txt

I then ran it again, unfortunately I plugged in the storage devices in the opposite order so I got a new copy of the SSD files put on the SD card, not a problem as this was just a test.

I then deleted the files on the SSD storage and the new entries on the SD card so they would be back to their original state and the state in which I would expect them to be in real life, picture files on the SD card and nothing on the SSD drive.

When I connected these to the Pi, following the prompts on my PC via the browser, external SSD storage in first and the SD card second.

The program started and stated it was comparing and gave the backup status. However this showed 11 of 1256 files and 0% with the time increasing from 00:9:25 to 3:21:50 before I stopped the program. I have attached the log file. Little-backup-box.log second use.txt

I then decided to perform a fresh creation of OS and download using imager etc.
I also re-formatted the SSD and SD card to make sure there was nothing hiding on these that might be confusing things.
I then ran LBB again, following the on screen instructions via the web browser on my PC. Connecting the SSD storage first and then the SD card.
Again the program seemed to hang, did comparison and then remained  at 10 of 982 and 1% adjusting time from 00:09:43 to 02:14:27 before I stopped the program.
The file number is lower as I loaded different files on the SD card, again to try to avoid anything that might remain on the card.
I have attached the log file.  Little-backup-box.log third use.txt

The configuration screen on my PC shows the log but is otherwise unresponsive, will not reboot or power down.

I am not sure what to try next.
The only change I think I mad in the settings is to activate power off after backup complete.

Any suggestions appreciated

David.

David Anderson

Little Backup Box
17:25:09
Little Backup Box

17:25:13
IP (online):
192.168.0.73
Little Backup Box

17:25:14
Source: storage

17:25:14
Destination: external 

17:25:30
Ready
Insert storage
IP (online):
192.168.0.73

17:28:09
DEVICE_IDENT_THIS='--uuid 3E63-F793' ('/dev/sde1') prepared to mount at '/media/storage'

17:28:11
mount_device RESULT=--uuid 3E63-F793

17:28:11
Ext. storage o.k.
size: 932G
free: 932G
Ready

17:28:15
Ready
Insert source
Ext. storage o.k.
[... 2241 lines cut ...]
2022/01/19 17:34:35 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1088.xmp
2022/01/19 17:34:37 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1089.NEF
2022/01/19 17:34:37 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1089.on1
2022/01/19 17:34:37 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1089.xmp
2022/01/19 17:34:38 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1090.NEF
2022/01/19 17:34:38 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1090.on1
2022/01/19 17:34:38 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1090.xmp
2022/01/19 17:34:40 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1091.NEF
2022/01/19 17:34:40 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1091.on1
2022/01/19 17:34:40 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1091.xmp
2022/01/19 17:34:41 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1092.NEF
2022/01/19 17:34:43 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1093.NEF
2022/01/19 17:34:44 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1094.NEF
2022/01/19 17:34:46 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1095.NEF
2022/01/19 17:34:47 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1096.NEF
2022/01/19 17:34:49 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1097.NEF
2022/01/19 17:34:50 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1098.NEF
2022/01/19 17:34:52 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1099.NEF
2022/01/19 17:34:53 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1100.NEF
2022/01/19 17:34:55 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1101.NEF
2022/01/19 17:34:56 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1102.NEF
2022/01/19 17:34:57 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1103.NEF
2022/01/19 17:34:59 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1104.NEF
2022/01/19 17:35:00 [10873] >f+++++++++ DCIM/2021-6-28 RAW/DSC_1105.NEF
2022/01/19 17:35:02 [10873]

outdoorbits, 2022-01-20:
Hi,

this is strange. I would have done the debugging just like you did. 
Unfortunately I didn't get a logfile of a broken backup-attempt. Could 
you provide one?

For me, there are a few ideas:

1. What format did you use for your ssd? I would suggest vfat.

2. Did any file arrive at the ssd? .id-file or files of the backup?

3. When backup seems to stop: Could it be, it is just working on a very 
large file? (Progress just counts the files created at the target, not 
the volume)

4. When backup seems to stop: Is /media/storage still mounted? (Use page 
System Info)

If Number 4 is "no": Looks like you are using a Nikon, probably XQD or 
CFExpress? There are lots of problems described with XQD, all about the 
power supply. Are you using a powered hub? This was the solution for me.

> Message ID: 
> ***@***.***>

dander45, 2022-01-20:
Hi Thanks for your reply,
I have just tried a powered USB hub and it appears to be working OK now.
This obviously increases the number of parts required to use this portable backup, PI, SSD drive, SD card reader, Powered USB Hub, battery pack.

What is the best solution for seeing what is going on during backup?
To be portable I won’t have a PC to use to connect to the device otherwise I could just use the PC to do the backup.
I have seen and it has been mentioned in this set of discussions, that a screen can be fitted is this simple and does LBB support the screen without modifications, or are there instructions of which screen to use and how to fit it?
Thanks for the help to now and hopefully the screen question as well

Thanks

David Anderson

outdoorbits, 2022-01-20:
Hi, have you visited the wiki? There are two very small hubs described. But it depends a lot on your power supply. Using my Anker 2700 mAh powerbank, XQD-backups are generally stable without a powered hub.
If you want to connect a display, please also have a look at the wiki. There is a description which type of display to use and how to connect it.
On my travels I used the display of lbb for default-mode-backups and my smartphone to start cloud-sync. I'm using comitup to connect my lbb.

closed 2022-02-28 by outdoorbits