outdoorbits/little-backup-box #133: Installation Process stuck at 90%

mrpaddel0815, 2022-01-14:
Hi i am trying to set up LBB - great Tool i think.
But i am stuck at the installation process. Got the raspi to run, ssh controll is active, 
but it gets stuck at 90% at "python3-networkmanager (2.2-1) ..."

Any ideas? Raspi 3B
[image]

outdoorbits, 2022-01-14:
Never seen that before. Did you try again? What size is your SD?

mrpaddel0815, 2022-01-14:
Hi, thank you for your quick reply. Yes did everything like your tutorial, got asked to set the mode i will backup and aked if i want to install 2 additional programs (not mentioned in the tutorial, but did it anyway). 32 GB SD and did it like 4 times now...

outdoorbits, 2022-01-14:
Hi, no idea. python-3-networkmanager seems to be downloaded at this point. Could there be a damage on your card? Could you try another one?

outdoorbits, 2022-01-14:
I just did a complete installation without any problems.

outdoorbits, 2022-01-14:
O.K., so this time the installation-process was successful? You can access by https or by http://IP:8000 I just added that to the installation-instructions. So if safari won't allow you to trust the certificate, you can use http at port 8000.

outdoorbits, 2022-01-14:
... chrome won' t allow you to trust the certificate ... ;-)

closed 2022-02-28 by outdoorbits