outdoorbits/little-backup-box #134: XQD card dont work with LBB

lgajewy, 2022-01-18:
Hello

I have a problem with run QXD cards.
It looks like the system sees the inserted card through the USB reader, but does not mount it either as storage (which would be an error, of course) or as SD.

I will only add that the SD cards work without a problem.

Have you done tests on QXD cards? (they are used, for example, in Nikon Z7 series)

(Rpi 4B, XDQ card reader, Samsung SSD disk as a storage)

BR,
LG

outdoorbits, 2022-01-18:
This seems to me to be a known problem - I had the same problem with nearly the same camera and maybe the same cards. But I guess, it's a hardware-problem about the power supply. Maybe it's working if you just use another (better) power adapter. In my case, I got stable conditions on my powerbank (Anker ~2700 mAh) or using a powered hub. Please have a look at https://github.com/outdoorbits/little-backup-box/wiki/02.-Hardware at the bottom of the file you will find Useful accessories and below that Travel hubs. I'm using the ORICO 4 Port USB3.0 Hub and since that I had no problems with XQD anymore.

lgajewy, 2022-01-18:
@outdoorbits - Bingo!
Big thx
When I used the original power supply, there is no problem with mounting the XQD card and copying the files. 
In first tests, I used the Platinet 20000 mAh (5V, 2 x 2.1A) powerbank - it is not enough.

closed 2022-02-25 by outdoorbits