outdoorbits/little-backup-box #141: Tiny file Manager won't show backup files because of ownership issues

VChoubard, 2022-02-03:
**Describe the bug**
When doing a backup to internal, the files backup are own by root:root instead of Apache (www-data:www-data). When you want to access it with tiny file manager they are not listed

**To Reproduce**
Steps to reproduce the behavior:
1. Do a backup with destination "internal"
2. Go to the web interface and open tiny file manager
3. Go to "internal" folder and then to the folder created
4. Observe that nothing is listed in the folder

**Expected behavior**
You can see the files that has been backup previously

**Desktop (please complete the following information):**
 - OS: Raspberry pi (buster)
 - Browser chrome

VChoubard, 2022-02-03:
I solve the issue by adding `sudo chown -R www-data:www-data "${BACKUP_PATH}"` in backup.sh line 564 but it is more a workaround than a proper fix.

outdoorbits, 2022-02-03:
Hmmm, the acl seems not to work correctly. I think, Your workaround is good for the moment but hope to find a more beautiful way by setfacl...

VChoubard, 2022-02-03:
The problem is the way we are accessing the Media folder, PHP `is_dir` and `is_file` are returning false on symlinks that don't have the same owner:groups.
A Proper solution will be not to use a symlink but tiny file manager only accept folder in the web directory (as far as I understand)

outdoorbits, 2022-02-03:
Strange, I can access my internal files by tinyfilemanager. Apache is configured to follow symlinks. The permissions are set by acl. This is done by the installation-process. Did you maybe delete and re-create the /media- subfolders? This could explain, why the acl are lost. Cloud you please update (just Settings - update) and tell me (even if it's working then again)?

closed 2022-02-25 by outdoorbits