outdoorbits/little-backup-box #145: Install Prozess stuck at 84% on python3-networkmanager (2.2-1) ...

Marown, 2022-02-23:
Hi,
i tried to install little backup box a view times, like in the tutorial, but still every time i got stuck at 84% on python3-networkmanager (2.2-1) ...

[image]

also the raspberry pi is no longer connecting to the network even after restarting so i have to put a new image on the SD Card to try again.

Marown, 2022-02-23:
Sry forgot
raspberry pi 3 B+
Sandisk Ultra 64GB C10

Marown, 2022-02-23:
Hi, i did the same thing as in https://github.com/outdoorbits/little-backup-box/issues/133
i tried to install it withou none prefered backup method but still it got stuck at 84%
[image]

Marown, 2022-02-23:
I found the Problem, i connected the pi with RJ45 instead of Wifi and also installed the lite OS 32 bit not the 64bit OS that worked for me

closed 2022-02-23 by Marown

Marown, 2022-02-23:
Yes I configured the wifi with the raspberry imager. Thanks for your awnser.

outdoorbits, 2022-02-24:
I tried a new installation on 64 bit lite on cable without any issues. Could you try again to reproduce the problem? Maybe it was a problem about the package or the SD card...?