outdoorbits/little-backup-box #146: SSD disconnects. Can't mount again

munecito, 2022-02-25:
**Describe the bug**
The Raspberry pi 4 B starts the backup but after some time the SSD disconnects and fails to mount again. The SSD appears to be there but can't mount it.

**To Reproduce**
Connect the SSD, connect the card reader and insert the Sd card

**Expected behavior**
To do a full backup

**Desktop (please complete the following information):**
Raspberry pi 4 B. 

Raspberry pi OS 64bit full (DESKTOP)

**Smartphone (please complete the following information):**
N/A

**Additional context**
The SSD is a sata type in a USB 3.0 enclosure. I suspect it is the pi ports that can't supply enough power to the enclosure. I have to try with an active hub but I have not got one.

If I do the backup to a Samsung T5 1Tb it does work. But I am not sure it would do a CF Express.

outdoorbits, 2022-02-25:
I think you are completely right. Most problems like this are caused by insufficient power-supply. I strongly recommend to use a powered usb-hub or at least a powerful power supply. Please have a look at [](https://github.com/outdoorbits/little-backup-box/wiki/02.-Hardware#useful-accessories)Useful accessories

outdoorbits, 2022-02-25:
https://github.com/outdoorbits/little-backup-box/wiki/02.-Hardware (Useful accessories)

closed 2022-02-28 by outdoorbits