outdoorbits/little-backup-box #157: External EXT4 HDD will not be mounted

Owz81, 2022-07-20:
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    0  1.8T  0 disk 
`-sda1        8:1    0  1.8T  0 part 
mmcblk0     179:0    0 59.6G  0 disk 
|-mmcblk0p1 179:1    0  256M  0 part /boot
`-mmcblk0p2 179:2    0 59.4G  0 part /

Owz81, 2022-07-20:
[image]

outdoorbits, 2022-07-20:
Would you please describe exactly, what you tried on LBB and how you got the output you posted?

Owz81, 2022-07-21:
Automated Backup Source to etx Storage.
Display says connect Storage. Than I connect this.
Display says Storage Size is 59GB ????
If I connet Source it will start Backup but wirting no Data to Storage.
I don't know were the data is wirten because, all den Data is gone after Restart.

If I mount the HDD via ssh, manual I can write and read from the HDD, no problem, but it doesn't happen automated via Applikation!

outdoorbits, 2022-07-21:
Would you please set loglevel to maximum, reboot for automated backup and send the logfile?
Also please connect your device, in tools do a checkdisk and if it detects an error, start a repair process.
Can you mount the device using tools? Can you access it by the filemanager then? Do you have read/write access?

Owz81, 2022-07-21:
Yes if I mount it manualy I can write and read files!
You will find some Logs from Backup Box and a ssh-log in the attachment.

[ssh.log](https://github.com/outdoorbits/little-backup-box/files/9159078/ssh.log)
[little-backup-box (2).log](https://github.com/outdoorbits/little-backup-box/files/9159079/little-backup-box.2.log)
[little-backup-box (1).log](https://github.com/outdoorbits/little-backup-box/files/9159080/little-backup-box.1.log)
[little-backup-box.log](https://github.com/outdoorbits/little-backup-box/files/9159081/little-backup-box.log)

outdoorbits, 2022-07-21:
Just reproduced the problem. It seems to be a problem associated to 
ext4. If you need a quick solution, just format your device as fat. I'll 
try to fix this problem, but this will maybe take a few days.

Owz81, 2022-07-21:
Ok i will try EXT3 and ExFAT the next days. I will let you know how it works, But a sulution for EXT4 would be nice!

Owz81, 2022-07-22:
Hmm with fat it reaks even more stange. ..
[little-backup-box (4).log](https://github.com/outdoorbits/little-backup-box/files/9171759/little-backup-box.4.log)
.

outdoorbits, 2022-07-22:
Please run an update (Settings - Update). I just added support for ext2, 
ext3, ext4, exfat and ntfs. My checks ran successfully.

Owz81, 2022-07-23:
I have the same issue with fat32 als befor. I will try a clean new setup today in the evening than we know more!

outdoorbits, 2022-07-23:
Please check also your partitions (use Tools). fat32 should not be a 
problem. What type of device are you using?

outdoorbits, 2022-07-23:
Without a log it is hard to guess the problem. ;-)

Owz81, 2022-07-23:
Log file 4 is the same as it behaved like bevor the Update. I will try a new install now!

Owz81, 2022-07-23:
OK now it works! There was no DCIM folder on the SD-Card just some picturs for try. I will make a try with EXT4 latest tomorrow!

Owz81, 2022-07-24:
Every thing works fine now. Many thank's to all contributors of this project and to autdoorbits for the fast fix!
I will write some lines in the discussions when I'm back from Namibia!

closed 2022-07-24 by Owz81