outdoorbits/little-backup-box #168: Oled display issue

clivetimmins, 2022-09-14:
Every other line on my oled display is missing, like it's being ran at 128x32 not 128x64. I have tried changing it in the python scripts as it seems that 32 is the default inside that ? Dosent seem to have changed anything. 
[image]

outdoorbits, 2022-09-14:
The display has four lines and it looks exactly like that. The last line is black now but I would expect, it was white when the display started. Just run a backup and I would expect you to see the use of the last line too.

outdoorbits, 2022-09-14:
New information is printed inverted, old lines are pushed down and printed as you see here in line 3. Outputs using all the screen (like on backup) do not follow this rule.

clivetimmins, 2022-09-15:
Shouldn't the white box for the inverted text be solid white? Instead of the zebra look it has now?  I should of been more clear I was referring to each line as in the individual pixel lines.

outdoorbits, 2022-09-15:
Does the last line work while backup?

clivetimmins, 2022-09-15:
Yes the last line works, just don't know why half the pixels are not showing up.
[image]

clivetimmins, 2022-09-15:
Thats right its only displaying 32 lines out of 64 I have counted to confirm. The adafruit library is set to 32 by default I think if that helps.

Default is
oled = adafruit_ssd1306.SSD1306_I2C(128, 32, i2c)

I think it should be 
oled = adafruit_ssd1306.SSD1306_I2C(128, 64, i2c)

Not sure if that helps at all. 
Thank you for taking the time on this I appreciate it.

clivetimmins, 2022-09-15:
I've figured it out. I was editing oled.py in my home folder not the var/www folder. 
[image]
[image]

closed 2022-09-15 by clivetimmins