outdoorbits/little-backup-box #172: Log monitor not working

caso1982, 2022-12-04:
I've just re-installed LBB and have noticed that it doesn't seem to be generating a .log file. 

[image]

outdoorbits, 2022-12-04:
Did you start any job before?

outdoorbits, 2022-12-04:
After boot lbb generates an empty logfile and sets the first entry: "Little Backup Box". Did you click on Refresh?

caso1982, 2022-12-04:
I'd tried starting a transfer job (USB - USB) and it works fine. There was no log file generated though. 
Also tried to refresh and download the log. No result from either.

outdoorbits, 2022-12-05:
Strange thing. Just to understand, This is a new installation on a newly 
installed latest version of Raspberry Pi OS Lite? Did you do any special 
configuration (including the settings of lbb)? Could you check for the 
logfile by

ls -la /var/www/little-backup-box/tmp/little-backup-box.log

just to check permissions.

caso1982, 2022-12-05:
This was an update that was overwritten over an older version of LBB (not sure which version) 
Output of ls -la /var/www/little-backup-box/tmp/little-backup-box.log is below. 

-rw-r--r-- 1 root root 337 Dec  5 20:19 /var/www/little-backup-box/tmp/little-backup-box.log

outdoorbits, 2022-12-05:
Strange, the file should be owned by user www-data and the 
install-/update-script should set this. Something went wrong, maybe the 
script didn't finish. We'll never know... Have fun!