outdoorbits/little-backup-box #173: No access after clean install

derekbez, 2022-12-18:
**Describe the bug**
Coming to Little-Backup-Box as a fresh new user.
After I follow the instructions to install on a RPI3, I connect my laptop (or my phone) to wifi network "little-backup-box-xxxx", and point my browser to 10.41.0.1.   All I get is Comitup "Choose WiFi Connection".  Am I missing a step to enable the AccessPoint?

**To Reproduce**
1. Find instructions at : https://github.com/outdoorbits/little-backup-box
2. Install as per instructions - 7 steps. 
3. Change laptop (or phone) wifi to "little-backup-box-xxxx".
4. Point browser to http://10.41.0.1 or https://10.41.0.1

**Expected behavior**
Expect to see little-backup-box.

**Install Log file**
```
pi@rpibackupbox:~ $ cat install-error.log
rm: cannot remove '/home/pi/little-backup-box': No such file or directory
Cloning into 'little-backup-box'...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  4669  100  4669    0     0  15773      0 --:--:-- --:--:-- --:--:-- 15827
Updating index cache for path `/usr/local/man/man1'. Wait...
done.
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.15.61-v7+
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
no crontab for pi
no crontab for pi
rm: cannot remove './tinyfilemanager': No such file or directory
Cloning into 'tinyfilemanager'...
Generating a RSA private key
........+++++
................................................+++++
writing new private key to '/etc/ssl/private/apache-selfsigned.key'
-----
--2022-12-18 16:41:30--  https://davesteele.github.io/comitup/latest/davesteele-comitup-apt-source_latest.deb
Resolving davesteele.github.io (davesteele.github.io)... 185.199.108.153, 185.199.109.153, 185.199.110.153, ...
Connecting to davesteele.github.io (davesteele.github.io)|185.199.108.153|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 43052 (42K) [application/octet-stream]
Saving to: ‘davesteele-comitup-apt-source_latest.deb’

     0K .......... .......... .......... .......... ..        100% 3.46M=0.01s

2022-12-18 16:41:30 (3.46 MB/s) - ‘davesteele-comitup-apt-source_latest.deb’ saved [43052/43052]

Configuration file '/etc/apt/sources.list.d/davesteele-comitup.list', does not exist on system.
Installing new config file as you requested.
Created symlink /etc/systemd/system/multi-user.target.wants/NetworkManager.service → /lib/systemd/system/NetworkManager.service.
Created symlink /etc/systemd/system/dbus-org.freedesktop.nm-dispatcher.service → /lib/systemd/system/NetworkManager-dispatcher.service.
Created symlink /etc/systemd/system/network-online.target.wants/NetworkManager-wait-online.service → /lib/systemd/system/NetworkManager-wait-online.service.
Unit dnsmasq.service does not exist, proceeding anyway.
Unit dhcpd.service does not exist, proceeding anyway.
Unit wpa-supplicant.service does not exist, proceeding anyway.
fusermount: failed to unmount /media/iOS: Invalid argument
fusermount: failed to unmount /media/cloud: Invalid argument
```

**Desktop (please complete the following information):**
 - Windows 11
 - Browser : Chrome / Edge
 - Version???

**Additional context**
I've tried this three times now - same every time.
RPi OS = 64bit Lite or 32bit Lite

FWIW, also found that the Raspberry Pi Imager (v1.7.3) allows Advanced Options > Configure wireless LAN and Enable SSH.  This means may not be necessary to manually create wpa_supplicant, etc.

Just an observation: in the instructions there is statement `Open a browser and enter https://IP.OF.YOUR.BOX`.  This might not be helpful to newbies.  Perhaps this could be changed to be explicit and more informative?

outdoorbits, 2022-12-18:
Hi,

you have been very close by your last lines. If you didn't connect your lbb to another wifi (by comitup), you can connect to your box just using this links:

https://10.41.0.1 (secure, certificate can't be verified automatically, please confirm it)
http://10.41.0.1:8000 (insecure)

If you connected it to another wify, you need to know the IP of your box in this wifi and use this IP instead of 10.41.0.1.

Did this work for you?
Stefan

derekbez, 2022-12-19:
Hi Stefan,

I deliberately do not connect to another wifi because the intention is to use this LBB out in the field where there is no wifi.

Whenever I go to 10.41.0.1 (http or https) I just get comitup asking me to connect to a wifi.   (Comitup shows a list of the available wifis around me at the moment.)

I've logged into the LBB console, and run ifconfig - it shows that the only IP available is 10.41.0.1.  (I only know very basic Linux, so perhaps this is not the best way to check?) 

Thanks

outdoorbits, 2022-12-19:
Maybe you browser does not accept the https-connection (because of an untrusted certificate)? Did you try http://10.41.0.1:8000/ (mind the :8000) also?

derekbez, 2022-12-19:
Okay - I think I have figured it out.
On the laptop, I can get to LBB using http://10.41.0.1:8000.  (I had only tried the port on the phone previously.)
On the phone (Android), I also have to turn off the mobile data connection.  Then I can get to LBB using the 10.41.0.1:8000.

It seems we always have to use the port :8000.  Without the port it just goes to Comitup.

Thanks for your help Stefan.  Hopefully my experience here can be documented into the wiki so as to perhaps help other users.

closed 2022-12-19 by derekbez