outdoorbits/little-backup-box #176: rclone GUI link goes to "404 page not found"

mrsparkle331, 2022-12-29:
**Describe the bug**
Attempting to access the rclone GUI leads to a "404 page not found"

Pi is connected to home wifi network.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Settings -> CONFIGURE CLOUD-SERVICES
2. Click on rclone GUI
3. New window opens to http://X.X.X.X:81/
4. See error - "404 page not found"

**Expected behavior**
I expect a web gui for rclone

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: macOS monterey 12.6 (21G115)
 - Browser Chrome 108.0.5359.124, Safari Version 16.0 (17614.1.25.9.10, 17614)

**Smartphone (please complete the following information):**
N/A

**Additional context**
LBB is current as of DEC 28 2022

outdoorbits, 2022-12-29:
I can't reproduce this actually but I hope to find the time for a new 
installation from scratch. I had a similar problem once but it 
disappeared after next reboot. Could you please perform an update (from 
config). This would try to install the gui again, just in case.

I just checked mine at http://x.x.x.x:81 as well as https://x.x.x.x:8443 
- both are working as expected.

Did you test this with another browser? Are you using a password for 
lbb?

Please tell me whether the problem persists after the update or not.

Message ID: ***@***.***>

mrsparkle331, 2022-12-30:
I just updated again and the logs show nothing was updated since everything was already up to date.

I tried https 8443 and http 81 and they both go to 404 not found.

I have a password set for lbb. It prompts for the password when I try to access the rclone GUI.

I used the two browsers listed in the original post.

outdoorbits, 2022-12-30:
Could you please have a look, if the server is running:

sudo ps auxw | grep rclone

Please post the output but overwrite the password. Should be something 
like

root         696  0.0  0.0  12168  4328 ?        S    09:50   0:00 sudo 
rclone rcd --rc-web-gui --rc-web-gui-no-open-browser --rc-addr :5572 
--config /media/rclone.conf --rc-user lbb --rc-pass XXXX
root         697  0.0  0.5 758760 42256 ?        Sl   09:50   0:00 
rclone rcd --rc-web-gui --rc-web-gui-no-open-browser --rc-addr :5572 
--config /media/rclone.conf --rc-user lbb --rc-pass XXXX
pi          9910  0.0  0.0   6040   636 pts/0    S+   10:06   0:00 grep 
--color=auto rclone

mrsparkle331, 2022-12-30:
`root         729  0.0  0.0  12168  4320 ?        S    02:20   0:00 sudo rclone rcd --rc-web-gui --rc-web-gui-no-open-browser --rc-addr :5572 --config /media/rclone.conf --rc-user lbb --rc-pass ******`

`root         730  0.0  0.5 759852 44616 ?        Sl   02:20   0:10 rclone rcd --rc-web-gui --rc-web-gui-no-open-browser --rc-addr :5572 --config /media/rclone.conf --rc-user lbb --rc-pass ******`

`pi         17738  0.0  0.0   6040   668 pts/0    S+   14:25   0:00 grep --color=auto rclone`

the password in the output matches the one I have set

outdoorbits, 2022-12-30:
Maybe I got it. Did you try to access the gui while the box was not connected to the internet? rclone downloads the gui on server-start. If internet is not connected at this time, it could be not available. Needs more investigation...

outdoorbits, 2022-12-31:
Please update, it should be there after that. To re-install/update the rclone-gui you can also use settings/Configure Cloud-services and check at "Update and restart rclone GUI"

mrsparkle331, 2023-01-03:
gui link is working after update! thanks!

closed 2023-01-03 by mrsparkle331