outdoorbits/little-backup-box #178: Log monitor and log files only contain time stamps

mrsparkle331, 2023-01-04:
**Describe the bug**
When viewing the log monitor in the browser there is no log output. you only see timestamps as the log updates.
When downloading the log file to view locally you see the same.

I did an update at jan 4 21:00 UTC and the behavior is still the same.

**To Reproduce**
Steps to reproduce the behavior:
open LBB webui.
View the log monitor.

**Expected behavior**
there should be log out put in the log monitor and log files.

**Log monitor and downloaded log file is essentially blank. It only contains time stamps:**

`Little Backup Box
21:22:43

21:22:46

21:22:46

21:22:46

21:22:46

21:22:46

21:22:46

21:22:46

21:22:46

21:22:46

21:22:46

21:22:46

21:22:47

21:22:47

21:22:47

21:22:47

21:22:57

21:22:57

21:23:02
`

**Desktop (please complete the following information):**
 - OS: mac os 13
 - Browser chrome 108.0.5359.124

mrsparkle331, 2023-01-04:
I have also tried changing the log level between 1,2, and 3. there is no difference

outdoorbits, 2023-01-04:
I just uploaded some changes a few minutes ago. Would you please update again? Most probably this does not affect your problem but makes debugging easier.

outdoorbits, 2023-01-04:
... I think I found the bug, it should be fixed now.

mrsparkle331, 2023-01-04:
working now!

closed 2023-01-04 by mrsparkle331