outdoorbits/little-backup-box #179: Display not working.

Nuramori, 2023-01-09:
I have an oled (128x64 display adafruit part https://www.adafruit.com/product/938) attached and properly wired up.  The display shows nothing.  Perhaps the I2C address needs changing, but I don't see where I can change that.

outdoorbits, 2023-01-09:
Could you please execute

sudo i2cdetect -y 1

on your pi and send the output? It should be something like

      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

In this example 3c would be the address.

> Message ID: ***@***.***>

Nuramori, 2023-01-09:
It says “3d”.   Sent from my iPhone

outdoorbits, 2023-01-09:
Could you please update, I did a little change, but it's not yet working out of the box.
Please login on your pi and edit /var/www/little-backup-box/oled.py
`nano +43 /var/www/little-backup-box/oled.py`
Change the line from
disp = adafruit_ssd1306.SSD1306_I2C(128, 64, i2c,addr=0x3C)
to
disp = adafruit_ssd1306.SSD1306_I2C(128, 64, i2c,addr=0x3D)

Does it work?

Nuramori, 2023-01-09:
> Anyway, I just added an option to config the i2c-address by settings: VIEW > i2c address of the display. Please tell me, does this work?

Sorry, I’ll do the update and testing tonight when I get back home.  Sent from my iPhone

Nuramori, 2023-01-10:
Yep, that got it to work!  Thanks.  Now I need to figure out how to get it to properly automatically mount usb devices I plug into the uugear four port hub.   Thoughts?Sent from my iPhone

outdoorbits, 2023-01-10:
I don't understand your question yet. If you face any problems, please just open a new issue.

Nuramori, 2023-01-10:
Btw I made a case for everything you may like that’s compact and is based on the newer pi zero W 2 combined with the oled I linked, with the uugear 4 port hub.  I can give more build info if you like it.  Sent from my iPhone

outdoorbits, 2023-01-10:
Great, I would be very interested in your case. Could you present it, maybe in  Discussions? https://github.com/outdoorbits/little-backup-box/discussions

closed 2023-01-10 by outdoorbits