outdoorbits/little-backup-box #180: "Little Backup Box Info:" email does not include IP address

mrsparkle331, 2023-01-11:
**Describe the bug**
No IP is shown in the start up info email

**To Reproduce**
Steps to reproduce the behavior:
1. set up email service
2. confirm test email works
3. shutdown device
4. start up device

**Expected behavior**
Email that is sent on start up contains no IP. See copy and paste of email:

> https (secure, certificate cannot be verified automatically, please confirm it):
> web UI: [https://](https:)
> 
> http (insecure):
> web UI: http://:8000/
> miniDLNA: http://:8200/
> 
> Open in file manager (Samba / Windows): smb://

**Desktop (please complete the following information):**
 - OS: macos
 - Browser: chrome
 - Version [e.g. 22] Version 108.0.5359.124

outdoorbits, 2023-01-11:
Could you please login at your lbb and enter to command
`hostname -I`
What does it repeat?

mrsparkle331, 2023-01-11:
I get the local IP of the pi

mrsparkle331, 2023-01-11:
I just did an update and the email has the IP now

mrsparkle331, 2023-01-11:
spoke too soon, I did another reboot and there was no ip in that email

mrsparkle331, 2023-01-11:
I narrowed it down a bit. The email with the IP was sent after the update was completed but before the reboot. after the reboot the email does not contain the IP.

Every time the update completes I get an info email that contains the IP, but not on a cold boot or normal reboot.

outdoorbits, 2023-01-11:
Did you do your update within the last hour? I changed the method to get the IP finally. 
Hard to say, why the IP is"empty", I never saw that issue.
I will add some temporary changes to the info mail to get closer to the problem.
Did you change anything in your installation (outside of LBB)?

outdoorbits, 2023-01-11:
The debug-update is online. Could you please post the debug-footer of a mail without IP?

outdoorbits, 2023-01-13:
Hi, I would like to remove the debug code. Did you already update and test?

mrsparkle331, 2023-01-14:
Hi sorry for the delay

i updated and tried again. The debug output shows the local ip

DEBUG:
ip route get 1:
1.0.0.0 via 192.168.8.1 dev wlan0 src 192.168.8.192 uid 0 cache 

hostname -I:
192.168.8.192

The email still shows blank ip except for the bb debug section

outdoorbits, 2023-01-14:
It looks like your connection has been unstable/changed just in the moment the mail is generated. LBB now waits for internet connection and IP. I think it's fixed.

mrsparkle331, 2023-01-14:
I see the ip in the email now! 

strange that there is a delay in getting ip. I have tried my pi 4 on three different networks in different locations and always got the blank ip. I just have a vanilla install. I am using desktop and not lite so maybe that’s causing the delay?

outdoorbits, 2023-01-14:
Maybe that's why I have never seen this before. :-) Can I mark this issue as closed?

mrsparkle331, 2023-01-14:
Yep closed

closed 2023-01-14 by mrsparkle331