outdoorbits/little-backup-box #183: Bug or stupid User?

ruebyi, 2023-01-12:
Hi, 

I'm not able to get the Email-Notification to work.
I tested
gmx.de
google.com
hosteurope

as Mailprovider - I don't receive mail and can't find any mails in the send Folders of this providers.

What am I doing wrong - I'm using the same settings as I use on my iPhone / iPad for those Accounts

outdoorbits, 2023-01-13:
Did this work before? I had to change a few things about mailing a few days ago. I'll test against GMX, but maybe not before Monday.
Could you set loglevel to max, mail security to STARTTLS, send a testmail and post the log? Thank you!

outdoorbits, 2023-01-13:
Does your password contain characters like ", ' or : ?

ruebyi, 2023-01-13:
It did work with an other Mailprovider, I'm not suing anymore.
But yes, my Passwords have some kind of special Characters - and the working one did not (Except for an "-")

ruebyi, 2023-01-13:
Here ist the Bug Report:

`15:40:16
OUTGOING MAIL
> curl --url 'smtps://smtp.gmail.com:465' --ssl-reqd --mail-from 'Lehmann.Tobias86@googlemail.com' --mail-rcpt 'tobias@mtlehmann.de' --user 'Lehmann.Tobias86@googlemail.com':'PASSWORD' --upload-file /var/www/little-backup-box/tmp/email.txt:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (67) Login denied

15:40:16
Mail:
Greetings from LBB

Congratulations, the mail delivery works.

`

ruebyi, 2023-01-13:
15:42:32
OUTGOING MAIL
> curl --url 'smtps://mail.gmx.net:587' --ssl --mail-from '*****@gmx.de' --mail-rcpt '*********' --user 'ruebyi@gmx.de':'PASSWORD' --upload-file /var/www/little-backup-box/tmp/email.txt:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (35) error:1408F10B:SSL routines:ssl3_get_record:wrong version number

15:42:32
Mail:
Greetings from LBB

Congratulations, the mail delivery works.

ruebyi, 2023-01-13:
I only deleted the Mailadressen and replaced them - sorry

outdoorbits, 2023-01-13:
I just updated lbb. This should make passwords more robust, only >'< 
inside a password should cause trouble. For mail.gmx.net I had success 
with port 465, STARTTLS. Does this work for you?

smu, 2023-01-13:
some problem here. I get a 
` curl: (67) Login denied`
without any special character is my password.

outdoorbits, 2023-01-13:
Did you give your username as number or mail-address?

smu, 2023-01-13:
I have my personal mailserver (exim) where I define username and password myself. 
No numbers and no '@' symbol or other special characters in the username or password.

ruebyi, 2023-01-13:
the fix worked for dmx - thank you!

smu, 2023-01-13:
In my case the problem must be related to my server 🙄. When using another mail server it works fine.

outdoorbits, 2023-01-13:
Does your servers mail logfile tell more?

smu, 2023-01-13:
Nope, only '535 Incorrect authentication data'

The verbose output of `curl` looks fine as well, including the base64 encoded username:password combination. I don't get it, but it is clearly not an issue in little-backup-box. thanks again!

Asturbike, 2023-01-14:
I think I only receive emails when the log level is maximum

outdoorbits, 2023-01-14:
You are right, thank you, bug is fixed.Message ID: 
***@***.***>

outdoorbits, 2023-01-17:
I'll close this issue now. Thank you for your bug reports.

closed 2023-01-17 by outdoorbits