outdoorbits/little-backup-box #184: memory issue or is the py3 (1GB RAM) not sufficient to run littel-backup-box?

smu, 2023-01-12:
At the moment pi4 models are hard to get so I though lets use one the pi3 models lying around to test your interesting software.
The installation worked fine, however, when importing photos from my camera the pi3 runs out of memory within minutes.

I know that a pi4 is recommended (due to the USB 3.0 ports). The pi3 has 1 GB of memory. Isn't this enough to run little-backup-box.

Below you can see a screenshot of bpytop, with shows how the systems runs out of memory..

[image]

outdoorbits, 2023-01-13:
I didn't check LBB for Pi 3 but there are reports, it's running on Pi zero. So I wouldn't expect problems in Pi 3. Did you install raspbian 32 bit light?

smu, 2023-01-13:
thanks for the answer and good to know. Yes, I installed raspbian 32 bit light.

Knowing that it should in general be possible to use a Pi 3, I will make a few more tests to understand this issue.

outdoorbits, 2023-01-13:
Maybe I have a Pi 3 somewhere, if yes, I'll check next week...

smu, 2023-01-13:
I think I found the problem. Either I have a  problem with the power supply or my external USB hard disc is broken. 
The PI runs out of memory, when the connection to the external drive is lost while importing photos.
Using another drive everything works fine. 

Sorry, I should have checked the log files for this yesterday. Now this issue can be closed as solved, i guess.
thanks for you support!

outdoorbits, 2023-01-13:
There have been some problems of power supply. Usually they are solved by a powered USB hub. For tested devices have a look at the wiki.

closed 2023-01-13 by outdoorbits