outdoorbits/little-backup-box #186: Cant get box to automount drives.

Nuramori, 2023-01-16:
No matter what I try, even reimaging the os and reinstalling the software, I can't get the device to automount any SD card nor storage device.  I've cleanly formatted the target device, and nothing seems to work.  I'm not sure where to start with diagnosing the problem.  Power is fine, and it even worked, once, but I can't get it to do it again.

outdoorbits, 2023-01-16:
What did you try? What is your source, a storage or a camera/smartphone? Did you setup a standard backup method? What if you start a backup USB to internal storage or another USB? Can you post the logfile?

Nuramori, 2023-01-16:
I am writing from an SD to portable HDD.  I am using the standard external source to target, and have tried multiple types of USB devices.  Nothing shows up as a mountable device.

outdoorbits, 2023-01-16:
You are using a RPi 4, raspbian lite?
Can you mount this devices using the same cables to a PC?

Nuramori, 2023-01-16:
I'm using a pi zero 2 W with the uugear 4 port usb hat. running raspian lite.  I can mount the devices on a PC.

outdoorbits, 2023-01-16:
I never tried but there are reports, it's working on this platform.
Trivial idea: USB and power are not reversed?
Could you log in (terminal), connect a storage device and check with lsblk ? Please post the message.
What's the filesystem type of your devices?

Nuramori, 2023-01-16:
sd is fat, hdd is exfat.  USB and power aren't reversed, as the system boots up fine.  Plugging in any USB results in no connection.

`NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 59.5G  0 disk
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 59.2G  0 part /
`

outdoorbits, 2023-01-17:
Your USB hub isn't powered, right? Most times things like that are a power problem. Do you see your SD card when HDD is disconnected?

Nuramori, 2023-01-17:
Issue resolved.  Fully user error.  The connecting pins were not touching the pads correctly.  Everything working as designed now - thanks!

closed 2023-01-17 by outdoorbits