outdoorbits/little-backup-box #188: Tranfer speed

Asturbike, 2023-01-22:
I am testing with the raspberry pi 4 and several hard disks and pendrives and i always find the same transfer speed.
Ex:
12:19:44
| Int. storage 
|  > USB storage 
| 9 of 115, 38.32MB/s
| Time: 00:00:47
| 7.8% >___________________

Have you exceeded 40 MB/s with any copy option or any special device?
I am currently using pendrive 3.0 and also SD card reader 3.0 with these results.
Maybe a certain pendrive or SSD option that I am not performing?
Thanks

outdoorbits, 2023-01-23:
I can confirm this. While USB 3.0 theoretically should make 625 MB/s possible, I get usually also ~40 MB/s. Best speed from internal to my SanDisk Extreme v1 was ~130 MB/s.

If I try a similar command locally on my laptop (using internal ssd nvme0n1p4):
`rsync -avh --info=FLIST0,PROGRESS2 --mkpath --no-perms --stats --min-size=1 ./tmp ./tmp2`
I also won't get much more than 280 MB/s. Any idea?

Asturbike, 2023-01-24:
I'm testing using an external ssd hard disk instead of micro sd

the operating system is faster, I will continue to report how lbb behaves.

Asturbike, 2023-01-29:
I currently have a raspberry pi 4 with the operating system on a ssd 2.0 disk and the transfer speed is:

dd if=/dev/zero of=test.img bs=1M count=1024 status=progress
1069547520 bytes (1.1 GB, 1020 MiB) copied, 25 s, 42.8 MB/s
1024+0 records in
1024+0 records out
1073741824 bytes (1.1 GB, 1.0 GiB) copied, 33.2713 s, 32.3 MB/s.

I tried to use my ssd disk with the 3.0 port of the raspberry but it does not boot, it only works when I connect it to the 2.0 port.

I am preparing a trip in which I need lbb to work perfectly so I will be all this month testing and configuring it as best as possible for me, thank you very much for your work.

outdoorbits, 2023-01-29:
But your SSD should work if you connect it to USB 3.0 after booting?

Asturbike, 2023-01-29:
I have a 3.0 hard drive enclosure but it only works plugged into the USB 2.0 port.
On other pcs with 3.0 port it works. 
I don't use sd card, the Raspberry Pi only uses the SSD to load the operating system.

outdoorbits, 2023-01-29:
I have no experience booting the Pi from USB devices. But I couldn't find a restriction regarding the USB 3.0 ports either...

Maybe I'm being overcautious, but a power outage or system crash at the wrong time could cause the system to become unbootable on the next start. In my countless tests I have had this situation several times - and in one case also on a trip.
Then a new installation was necessary. If the data and the system are on the same medium, a new installation also costs the data.
Based on this consideration, I always take at least two fully installed SD cards with me when traveling. Last summer in Greece, that was my luck!

closed 2023-03-05 by outdoorbits