outdoorbits/little-backup-box #190: Not asked to install Comitup - no service

manni0815held, 2023-02-04:
I installed the little backupupbox as told. The installation seems to work fine, except i didn't get asked to install comitup.
After the installation the connection didnt break up. And there is no Comitup Wifi acces point. 
But there are folders with littlebackupbox and tinyfilemanager on the raspberry. But if i try to backup sd cards nothing happens.

[image]

Can anyone help me? I have no idea what im doing or doing wrong :(

Thank you

outdoorbits, 2023-02-04:
Would you please provide more information? Pi 4, Raspberry Pi OS Lite 32 or 64?

manni0815held, 2023-02-04:
Pi 4 4gb. Tried both lite 32 and 64, at the moment 64.

outdoorbits, 2023-02-04:
I just started an installation and everything worked find up to now. I was asked about standard backup method and comitup, everything as expected up to now. You got only one question?

outdoorbits, 2023-02-04:
How did you try to start your backup?

manni0815held, 2023-02-04:
No, i got no questions at all.
And i just connected a T5 1TB SSD and then a SD card Reader with sd card. After a while i took the SSD and had a look if anything was copied.

outdoorbits, 2023-02-04:
Ok, you obviously didn't define a standard backup method? Did you run the installation script as mentioned? How did you install LBB?

manni0815held, 2023-02-04:
I installed the image via raspberrypi imager. Defined my wifi before that so i could ssh the raspberry. With ssh connection i pasted "curl -sSL https://raw.githubusercontent.com/outdoorbits/little-backup-box/main/install-little-backup-box.sh | bash" in the commandline. I took 20 minutes with no feedback or questions and ended.

outdoorbits, 2023-02-05:
I just did the same:

By imager I selected "RASPBERRY PI OS LITE (64-BIT), enabled ssh, set a new password for user pi and defined my wifi.
Now I booted this image on my pi and logged in `ssh pi@xxx.xxx.xxx.xxx`.

A few seconds after ` curl -sSL https://raw.githubusercontent.com/outdoorbits/little-backup-box/main/install-little-backup-box.sh | bash  2> install-error.log` I got this screens:

[image]
[image]

That's just what it should do.

If you just did the same, there should be equal conditions on both of our raspberry pi's. Maybe your client isn't able to display the dialog? What ssh client are you using?

outdoorbits, 2023-02-05:
Please ssh to your pi and try this:
`dialog --yesno "test" 5 30`
Does this display a blue dialog?

manni0815held, 2023-02-05:
Ok, i used WINSCP, that was the problem. Now it worked with putty, the dialog came up.
There is a littlebackupbox Wifi now, but i only can see a site where it says "Choose wifi".

outdoorbits, 2023-02-05:
Did you try this?

*** Alternatively can reach the web UI of Little Backup Box when you are connected to its
*** hotspot as follows:
***
*** https://10.41.0.1 (secure, certificate can't be verified automatically, please confirm it)
*** http://10.41.0.1:8000 (insecure)

manni0815held, 2023-02-05:
I connected to the littlebackupbox wifi. and tried the ip's.
But it just opens a screen where i have to choose one of my own wifis, nothing else.
On PC and on Iphone.

[image]

manni0815held, 2023-02-05:
Ok, it worked. With the 8000 Port.
Thank you so much. Do i have to close this issues by myself?

outdoorbits, 2023-02-05:
But if you configure one of these wifis, your lbb will connect to it now and next times. You just have to find out your new IP. This is easier if using a display.
Did you mention, there are differences between http and https? (README)

closed 2023-02-05 by outdoorbits