outdoorbits/little-backup-box #204: Comitup isn't instralled correctly during setup

Master-Mace, 2023-05-05:
**Describe the bug**
During setup of LLB, the 'Comitup' package isn't installed correctly. Tried it twice and after examining the log files, it seems that the packages needed are no longer available.

**Log File**
--2023-05-05 16:59:25--  https://davesteele.github.io/comitup/latest/davesteele-comitup-apt-source_latest.deb
**Resolving davesteele.github.io (davesteele.github.io)... failed: Name or service not known.**
wget: unable to resolve host address ‘davesteele.github.io’
ESC[1mdpkg:ESC[0m ESC[1;31merror:ESC[0m cannot access archive 'davesteele-comitup-apt-source_latest.deb': No such file or directory
rm: cannot remove 'davesteele-comitup-apt-source_latest.deb': No such file or directory

**Solution**
Package seems to have moved to:
davesteele-comitup-apt-source*.deb
Please update the installation script with this.

outdoorbits, 2023-05-05:
Thank you very much! I just asked the maintainer of comitup what path to follow. I will fix this as soon as possible.

Master-Mace, 2023-05-05:
Great, thanks! Glad to help out, great piece of software!!!

outdoorbits, 2023-05-09:
I checked all the process again and found a confusing link. But asking the developer of comitup, everything is correct and in place. The test this, you can ssh to your lbb and ...

`cd little-backup-box/`
`sudo ./install-comitup.sh`

It works fine here. Can you still reproduce the error?

Master-Mace, 2023-05-10:
I had to reinstall LBB onto a new micro-sd card and ran the script from the install page and this time Comitup installed directly. Seems to be fixed, can be closed.

closed 2023-05-10 by outdoorbits